chipsalliance/synlig #554: Run some type of function will meet error

little-blue, 2022-03-11:
I use the uhdm to translate the orv64 SystemVerilog file into the verilog file, and want to do synthesis by yosys, the orv64 gitlab adress is here:  [https://gitlab.com/picorio/OpenSource/hardware/orv64](url)
But there is some error when doing this translate , when the function include following format(Variables include output)， the code where use this function will meet error. Here is an example in the orv64 
```
function automatic void lru8_update_lru(
  output logic [6:0] new_lru,
  input logic [6:0] old_lru,
  input logic [2:0] way_id
);
  new_lru = old_lru;
  new_lru[6] = ~old_lru[6];
  if (way_id[2])
    new_lru[5] = ~old_lru[5];
  else
    new_lru[4] = ~old_lru[4];

  new_lru[way_id[2:1]]  = ~old_lru[way_id[2:1]];

endfunction
```

and will meet this error
[image]

the function screenshot is this 
[image]

what we can do is change the source code where use this function, and expand this function where it is used, and this problem will be sovled.

rkapuscik, 2022-03-16:
Thank you for reporting this, I have reproduced the error and we will investigate it.

mithro, 2022-03-24:
This issue is closed, was it fixed?

rkapuscik, 2022-03-25:
I've just retested with current plugin, the issue is still present. Reopening.