chipsalliance/synlig #555: Can‘t find width for identifier

little-blue, 2022-03-11:
I would like to translate the sv code of orv64 [(https://gitlab.com/picorio/OpenSource/hardware/orv64)](url) into verilog code and use yosys to synthesis it.
and I meet some error uhdm can't find width for identifier. Here is an example in orv64
the definition of a varible is like
`typedef logic [ORV64_CNTR_WIDTH-1:0]        orv64_cntr_t;`
and in other place need the width of this varible, so there is a refer code: by using the 
`$bit(orv64_cntr_t)-1:0` like:
`if(ma2cs.csr_addr == ORV64_CSR_ADDR_MHPMCOUNTER3) hpmcounter3 <= csr_wdata[$bit(orv64_cntr_t)-1:0];`
will meet

 error like this screenshot:
[image]

we can only manully replace the width where use the `$bit(orv64_cntr_t)-1:0` to avoid this error as follow:
`if(ma2cs.csr_addr == ORV64_CSR_ADDR_MHPMCOUNTER3) hpmcounter3 <= csr_wdata[ORV64_CNTR_WIDTH-1:0];`

here is the whole code screenshot, the file reference is all already

[image]

[image]

rkapuscik, 2022-03-16:
Thanks for your report and the test case. I have reproduced the error. It is caused by inability to resolve an imported symbol and is similar to an error we encountered in our test suite.

We will look into that.

mithro, 2022-03-24:
This issue is closed, was it fixed?

rkapuscik, 2022-03-25:
I've just retested with current plugin, the issue is still present. Reopening.

mglb, 2022-10-24:
Probably the same issue: https://github.com/antmicro/yosys-systemverilog/issues/1064