chipsalliance/synlig #582: Segementation fault

little-blue, 2022-03-20:
No additional information
I would like to translate the sv code of orv64 into verilog code and use yosys to synthesis .[orv64](https://gitlab.com/picorio/OpenSource/hardware/orv64)
Here is a test case, I just use the orv64_execute as a top
(latter update)

rkapuscik, 2022-03-22:
Thank you for your report. With current frontend version, I don't see a segmentation fault for your test case, but `ERROR: Abort in frontends/ast/genrtlil.cc:705.`, so the generated AST is incorrect.

We will investigate this further.

rkapuscik, 2022-03-25:
The segfault is gone and it looks like the error is the same as https://github.com/antmicro/yosys-uhdm-plugin-integration/issues/554, so it will be tracked there.

closed 2022-03-25 by rkapuscik