chipsalliance/synlig #649: Run yosys and plugin with address sanitizer

hzeller, 2022-04-11:
Running the tests with address sanitizer ( clang doc: https://clang.llvm.org/docs/AddressSanitizer.html ; should be similar for gcc/clang) will help finding some issues early.

It does result in somewhat slower binaries (but _much_ less so than e.g. valgrind), so the CI machines will have to work harder, but any findings here will well offset the engineering time attempting to find a head-scratching issue later.

hzeller, 2022-07-15:
I think this gets more important as we can find crashes that might not be detected otherwise (like the use-after-delete situation in https://github.com/chipsalliance/yosys-f4pga-plugins/issues/366)

hzeller, 2022-07-15:
I don't have permission to assign owners to bugs in this repo, so CC: @rkapuscik @tgorochowik

mglb, 2023-05-17:
Possible blocker: https://github.com/chipsalliance/Surelog/issues/3661

I had some troubles in making ASAN work due to lack of support for code loaded with dlopen/dlclose, but I was able to use it after all with the help of this workaround: https://github.com/google/sanitizers/issues/89#issuecomment-406316683