chipsalliance/synlig #672: Surelog version ancient

hzeller, 2022-04-18:
The Surelog version [currently refernced in the submodule](https://github.com/chipsalliance/Surelog/tree/5c2bac5b95fc1d08f8b1b44a19391030ef069a02) is over a month old and it seems that the latest UHDM module does not work anymore with the current Surelog (still verifying).
Also Yosys is 10 days old.

This means the current module can't be used from head as users have to juggle various potentially different versions to figure out which combination is working.

We should at max be 2-3 days behind head of any given dependency, with aim for < 1 day.
It should be possible to compile, build and test with all dependencies from head at least once a day.

pgielda, 2022-04-18:
Seems there is a dependabot PR #670 but CI is red?

pgielda, 2022-04-18:
CI is read because of compilation issues:
`07:23:41 | UhdmAst.cc: In member function 'Yosys::AST::AstNode* Yosys::UhdmAst::process_object(vpiHandle)':
07:23:41 | UhdmAst.cc:3924:16: error: 'uhdmimport' is not a member of 'UHDM'; did you mean 'uhdmport'?`

hzeller, 2022-04-18:
Yep, filed that here https://github.com/chipsalliance/yosys-f4pga-plugins/issues/303

hzeller, 2022-04-18:
But that changed only a few days ago and wouldn't explain a month old Surelog.
If dependabot was running, maybe it silently did not update things as it started to fail a while ago ?

pgielda, 2022-04-18:
There is also #612 that seems to be some WIP on updating Surelog so I guess there is more issues to be fixed.

hzeller, 2022-04-19:
I think somewhere between Surelog 7e0222abdca616cc0b18a471efe7a73d1f6c30fb cd26de17ab2474cc6bc4ff54aad97a495b8f3bc2 things didn't work anymore with the module.

hzeller, 2022-04-22:
BTW, I've patched in the change yesterday and it seems to work with current head of Surelog.
Did y ou narrow down on the parameter override problem you were seeing ?

rkapuscik, 2022-04-22:
> Did you narrow down on the parameter override problem you were seeing ?

Reported here: https://github.com/chipsalliance/Surelog/issues/2881. Plugin makes sure there are no errors from Surelog before reading UHDM, so it breaks the tests now.

closed 2023-12-07 by alaindargelas