chipsalliance/synlig #686: Encountered unhandled object '' of type 'break_stmt'

hello-eternity, 2022-04-24:
I got an error like the title. The detailed error information is like this 
`ERROR: /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests/orv64/rtl/common/pygmy_func.sv:24: Encountered unhandled object '' of type 'break_stmt'`
The source code is :
```
    for (int i=0; i<8; i++) begin
      if (~way_valid[i] & way_enable[i]) begin
        has_empty_way = '1;
        replace_way_id = i;
        break;
      end
    end

```
I can only sovle this by change the for-loop logic 
This is the test case “https://github.com/hello-eternity/test_orv64_uhdm”  
In the Makefile.in , $(TEST_DIR) need to 
 point to the correct location. Mine is `/root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests`
This error need replace this file `orv64/rtl/common/pygmy_func copy.sv` into `orv64/rtl/common/pygmy_func.sv`

hello-eternity, 2022-06-19:
CC @hzeller

kbieganski, 2022-06-22:
`break` statements aren't supported yet. They are a work in progress: https://github.com/antmicro/yosys-f4pga-plugins/tree/sv-plugin-break-continue

We'll let you know once it's finished.

kbieganski, 2022-06-29:
`break` and `continue` support landed in mainline: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/357
Let us know if it worked for you once you get a chance to test it!

renu883, 2023-08-30:
facing the same issue but with function calls.

renu883, 2023-08-30:
steps to reproduce:
read_systemverilog -defer f1.sv
read_systemverilog -defer f2.sv
...
read_systemverilog -defer f22.sv
read_systemverilog -link