chipsalliance/synlig #687: UHDM coverage  point to empty source line

hello-eternity, 2022-04-24:
I got an error like the title. The detailed error information is like this 
[image]

I should delete the empty source line to fix it.
This is the test case “https://github.com/hello-eternity/test_orv64_uhdm”  
In the Makefile.in , $(TEST_DIR) need to 
 point to the correct location. Mine is `/root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests`
This error need replace this file `orv64/subproj/es1y/rtl_gen/station_vp copy.sv` into `orv64/subproj/es1y/rtl_gen/station_vp.sv` to reproduce.

rkapuscik, 2022-04-28:
Thanks for reporting this! This error (and all other messages preceded by square brackets) actually come from [Surelog](https://github.com/chipsalliance/Surelog). Can you report it there?

hello-eternity, 2022-04-29:
OK, Here will be closed