chipsalliance/synlig #688: Segmentation Fault

hello-eternity, 2022-04-24:
I use the latest uhdm to run this orv64 code and encounter segementation fault without more information
[image]

This is the test case “https://github.com/hello-eternity/test_orv64_uhdm”  
In the Makefile.in , $(TEST_DIR) need to point to the correct location. Mine is
 `/root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests`

hello-eternity, 2022-04-29:
I run it again and still segmentation fault. You may not change back the some copy file after other error test. I upload a new branch named segmentation, you can dirctly download this branch to test.

hello-eternity, 2022-06-19:
CC @hzeller

alaindargelas, 2022-07-31:
@hello-eternity, I'm trying to reproduce this issue.
Please write down the exact commands to reproduce the issue (No English description as above, it is too vague), please just list all the commands:

clone https://github.com/hello-eternity/test_orv64_uhdm.git
cd test_orv64_uhdm/
???