chipsalliance/synlig #689: Encountered unhandled typespec in process_typespec_member: 'orv64_access_type_t' of type 'unsupported_typespec'

hello-eternity, 2022-04-24:
The full error is
 `ERROR: /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests/orv64/subproj/es1y/rtl_gen/pygmy_intf_typedef.sv:147: Encountered unhandled typespec in process_typespec_member: 'orv64_access_type_t' of type 'unsupported_typespec'`

the source code definition is 
```
typedef struct packed {
  orv64_vpn_t req_vpn;
  orv64_access_type_t req_access_type;
} orv64_tlb_ptw_if_req_t;
```
where orv64_access_type_t is 
```
  typedef enum logic [1:0] {
    ORV64_ACCESS_FETCH = 2'b00,
    ORV64_ACCESS_LOAD  = 2'b01,
    ORV64_ACCESS_STORE = 2'b10,
    ORV64_ACCESS_AMO   = 2'b11
  } orv64_access_type_t;
```

the test case is [https://github.com/hello-eternity/test_orv64_uhdm/tree/typebug](https://github.com/hello-eternity/test_orv64_uhdm/tree/typebug)

rkapuscik, 2022-04-28:
Thanks for your continued interest in the plugin! The `unsupported typespec` should not be present in the UHDM, this indicates an error in Surelog (which is used to parse the design) when it cannot resolve the type. This could mean that the design is set up incorrectly or a bug in Surelog. Could you verify if there are errors or warnings in `build/slpp_all/surelog.log`?

hello-eternity, 2022-04-29:
No
`[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 619
[   NOTE] : 166
cp /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/build/slpp_all/surelog.uhdm /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/build/orv64.uhdm
(cd /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/build && /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/../image/bin/yosys -s /root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests/orv64//yosys_script)

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.16+41 (git sha1 29c0a5958, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

-- Executing script file `/root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests/orv64//yosys_script' --

!! DEPRECATION WARNING !!

The uhdm plugin has been renamed to systemverilog.
Loading the systemverilog plugin...

1. Executing UHDM frontend.
^Cmake: *** [Makefile:56: uhdm/yosys/test-ast] Interrupt`
In fact there is no error and continue to the next step and met the following error

hello-eternity, 2022-06-19:
CC @hzeller

alaindargelas, 2022-07-31:
If I run the following project file in Surelog standalone:

```
+define+PYGMY_ES1Y+FPGA+STATION_VP_PKG__SV+USE_POWER_PINS  
    rtl/common/pygmy_define.sv  
    rtl/common/pygmy_cfg.sv  
    rtl/common/pygmy_typedef.sv  
    rtl/common/pygmy_func.sv  
    rtl/png-32/rtl/orv/orv_cfg.sv 
    rtl/png-32/rtl/orv/orv_typedef.sv 
    rtl/png-32/rtl/orv/orv_func.sv 
    rtl/plic/plic_typedef.sv 
    rtl/orv64/orv64_param_pkg.sv 
    rtl/orv64/orv64_typedef_pkg.sv 
    rtl/orv64/orv64_func_pkg.sv 
    rtl/vcore_gen2/rtl/vcore_cfg.sv 
    rtl/vcore_gen2/rtl/vcore_pkg.sv 
    rtl/common/oursring_typedef.sv 
    subproj/es1y/rtl_gen/pygmy_intf_typedef.sv 
    subproj/es1y/rtl_gen/orv64_csr_mmu_if.sv 
    subproj/es1y/rtl_gen/orv64_dc_ma_if.sv 
    subproj/es1y/rtl_gen/orv64_ex_dc_if.sv 
    subproj/es1y/rtl_gen/orv64_tlb_flush_if.sv 
    subproj/es1y/rtl_gen/oursring_req_if.sv 
    subproj/es1y/rtl_gen/pcie_req_if.sv 
    subproj/es1y/rtl_gen/usb_cache_if.sv 
    subproj/es1y/rtl_gen/vpfp_chkr_csr_if.sv 
    subproj/es1y/rtl_gen/station_dma.sv 
    subproj/es1y/rtl_gen/mem_bar_status_if.sv 
    subproj/es1y/rtl_gen/pcie_resp_if.sv 
    subproj/es1y/rtl_gen/spi_m_req_if.sv 
    subproj/es1y/rtl_gen/station_dt.sv 
    subproj/es1y/rtl_gen/l2_ctrl_cfg_if.sv 
    subproj/es1y/rtl_gen/mem_bar_clear_banks_if.sv 
    subproj/es1y/rtl_gen/spi_s_rsp_if.sv 
    subproj/es1y/rtl_gen/station_ddr_top.sv 
    subproj/es1y/rtl_gen/orv64_if_ic_if.sv 
    subproj/es1y/rtl_gen/station_usb_top.sv 
    subproj/es1y/rtl_gen/dma_intr_if.sv 
    subproj/es1y/rtl_gen/dma_data_avail_if.sv 
    subproj/es1y/rtl_gen/scan_f_if.sv 
    subproj/es1y/rtl_gen/dtm_if.sv 
    subproj/es1y/rtl_gen/tcm_if.sv 
    subproj/es1y/rtl_gen/station_orv32.sv 
    subproj/es1y/rtl_gen/station_slow_io.sv 
    subproj/es1y/rtl_gen/mem_bar_status_banks_if.sv 
    subproj/es1y/rtl_gen/vpfp_chkr_vrf_if.sv 
    subproj/es1y/rtl_gen/station_pll.sv 
    subproj/es1y/rtl_gen/apb_if.sv 
    subproj/es1y/rtl_gen/station_vp.sv 
    subproj/es1y/rtl_gen/jtag_if.sv 
    subproj/es1y/rtl_gen/orv64_cache_tlb_if.sv 
    subproj/es1y/rtl_gen/vpfp_chkr_irf_if.sv 
    subproj/es1y/rtl_gen/ahb_if.sv 
    subproj/es1y/rtl_gen/orv64_tlb_ptw_if.sv 
    subproj/es1y/rtl_gen/oursring_resp_if.sv 
    subproj/es1y/rtl_gen/io_boot_if.sv 
    subproj/es1y/rtl_gen/vpfp_chkr_wb_if.sv 
    subproj/es1y/rtl_gen/station_byp.sv 
    subproj/es1y/rtl_gen/station_cache.sv 
    subproj/es1y/rtl_gen/acp_if.sv 
    subproj/es1y/rtl_gen/amo_store_if.sv 
    subproj/es1y/rtl_gen/vpfp_chkr_id_if.sv 
    subproj/es1y/rtl_gen/test_io_if.sv 
    subproj/es1y/rtl_gen/cache_mem_if.sv 
    subproj/es1y/rtl_gen/station_sdio.sv 
    subproj/es1y/rtl_gen/cache_fence_if.sv 
    subproj/es1y/rtl_gen/cpu_cache_if.sv 
    subproj/es1y/rtl_gen/csr_satp_if.sv 
    subproj/es1y/rtl_gen/station_test.sv 
[... 876 lines cut ...]

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:231:26: Unresolved hierarchical reference "ring_req_if_w.wdata".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:232:22: Unresolved hierarchical reference "ring_req_if_aw.awid".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:164:10: Unresolved hierarchical reference "ring_resp_if_b.bresp".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:165:10: Unresolved hierarchical reference "ring_resp_if_r.rresp".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:166:10: Unresolved hierarchical reference "ring_resp_if_b.bid".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:167:10: Unresolved hierarchical reference "ring_resp_if_r.rid".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:168:10: Unresolved hierarchical reference "ring_resp_if_r.rdata".

[ERR:UH0725] rtl/orv64/orv64_debug_access.sv:169:10: Unresolved hierarchical reference "ring_resp_if_r.rlast".

[INF:UH0708] Writing UHDM DB: ./slpp_all/surelog.uhdm ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 207
[WARNING] : 562
[   NOTE] : 99
```