chipsalliance/synlig #768: Encountered unhandled typespec in process_parameter:  of type  short_int_typespec

gkamendje, 2022-05-21:
I have modified the OpenROAD flow scripts to use the systemverilog plugin to read SV files. 
The line `parameter shortint unsigned W_LFSR = 16;` in my file `my_pkg_defs.sv` causes YOSYS to issue an error during synthesis

```
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 127
[   NOTE] : 5
[00000.568422] ERROR: ../rtl/my_pkg_defs.sv:12: Encountered unhandled typespec in process_parameter: '' of type 'short_int_typespec'

```
No error is reported in the file slpp_all/surelog.log. So I am not sure it this is a YOSYS issue or a SV plugin issue.

rkapuscik, 2022-05-23:
Thanks for reporting this, this is a plugin error. It should be fixed by https://github.com/chipsalliance/yosys-f4pga-plugins/pull/332.

gkamendje, 2022-05-25:
Thanks a lot. I will grab the next release.

closed 2022-05-25 by gkamendje