chipsalliance/synlig #938: Formal equiv diff simple_func_width_scope.v

alaindargelas, 2022-07-28:
This test is a trimmed down/modified version of yosys/tests/simple/func_width_scope.v
The UHDM tree seems all correct, yet there is a formal verification diff in the gate output. It looks like the plugin is setting func1[0] == ~inp and func1[1-5] == 1, where the correct answer would be func1[0-5] == ~inp

```
module func_width_scope_top(inp,xn);
	localparam WIDTH_A = 5;
	input wire signed inp;
   	output reg [WIDTH_A-1:0] xn;

	function automatic [WIDTH_A-1:0] func1;
		input reg [WIDTH_A-1:0] inp;
		func1 = ~inp;
	endfunction

	assign xn = func1(inp);

endmodule
```

[image]

mandrys, 2022-11-21:
Needed issue https://github.com/chipsalliance/Surelog/issues/3346 to be solved too.

mandrys, 2022-12-06:
The originally spotted issue with _simple_func_width_scope.v_ presented in the trimmed down version of the test could be solved by the fix from https://github.com/chipsalliance/Surelog/issues/3339. However, the full _simple_func_width_scope.v_ failed the formal verification again, and it was caused by a wrong return value from the func call inside the range. It should be fixed by now in Surelog, https://github.com/chipsalliance/Surelog/issues/3346, but fixing it in the plugin is still work in progress.

In the meantime, an issue with the return values from the func call were spotted for another test: https://github.com/YosysHQ/yosys/blob/master/tests/simple/const_func_shadow.v