chipsalliance/synlig #949: Can't load module ./systemverilog: libreadline.so.7

jeras, 2022-07-31:
I updated the plugin to the latest version 4199cbc-2022-07-29 while yosys is also from latest git.
I got the next error, which I think is a regression introduced in the last few days.

```
$ yosys -s build.tcl 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.19+36 (git sha1 30a4218f5, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

-- Executing script file `build.tcl' --
ERROR: Can't load module `./systemverilog': libreadline.so.7: cannot open shared object file: No such file or directory
```

I am using Ubuntu 20.04 and the available library is
```
libreadline-dev/focal,now 8.0-4 amd64 [installed]
  GNU readline and history libraries, development files

libreadline8/focal,now 8.0-4 amd64 [installed,automatic]
  GNU readline and history libraries, run-time libraries
```

jeras, 2022-07-31:
I installed the plugin from the release tarball:
https://github.com/antmicro/yosys-uhdm-plugin-integration/releases/download/4199cbc-2022-07-29/yosys-uhdm-plugin-4199cbc-Ubuntu-20.04-focal-x86_64.tar.gz

And I compile yosys myself.

alaindargelas, 2022-07-31:
@jeras, please don't use the tarball. It does not work for me most of the time.
Instead, can you build all the executable from scratch in a systematic way:
cd yosys-uhdm-plugin-integration
git pull
./build_binaries.sh --build-sv2v
./install_plugin.sh    or    make sure you point your yosys to yosys-uhdm-plugin-integration/yosys/yosys

jeras, 2022-07-31:
Thanks for warning me against using the tarball.
I rebuilt the plugin project, the with one extra git step:
```
cd yosys-uhdm-plugin-integration
git pull
git submodule update --recursive
./build_binaries.sh --build-sv2v
```

I would prefer to use the executables from `yosys-uhdm-plugin-integration/image/bin`, to avoid installing.
In addition to the `PATH` variable, do I have to set up any other environment variables, for example `LD_LIBRARY_PATH` for the `*.so` files? I do not know if `image/share/yosys` files will be used by `image/bin/yosys` or will it use files installed on the system.

alaindargelas, 2022-08-01:
I think you only have to setenv LD_LIBRARY_PATH to image/share/yosys/plugins/

yosys-uhdm-plugin-integration$ ls image/share/yosys/plugins/
design_introspection.so  fasm_extra_modules/      integrateinv.so          ql-iob.so                sdc.so                   uhdm.so                  
dsp-ff.so                fasm.so                  params.so                ql-qlf.so                systemverilog.so         xdc.so

alaindargelas, 2022-08-02:
@tomaszgielda, please check in the plugin code why there is this error condition.
I'll check the UHDM tree.

alaindargelas, 2022-08-05:
@jeras , please close this issue if it is fixed.

jeras, 2022-08-06:
Well I did not debug this far, and it always seemed more like something I did wrong.
I will close it now an reopen it if I encounter it again.

closed 2022-08-06 by jeras