chipsalliance/synlig #967: uhdm plugin misses one optimization pass

alaindargelas, 2022-08-06:
When comparing gate level netlists generated by the Yosys parser and UHDM, it looks like the UHDM plugin is missing an optimization pass for non-observable logic.

sv2v/test/relong/split_ports.v

./run_formal_verif.sh test= relong_split_ports.v

cd build/tests/relong_split_ports.v

tkdiff yosys_gate.v surelog_gate.v

Surelog contains an instance that should have been optimized away as it is unobservable.
Either are are not passing some properties to the Yosys AST, or we are missing a call for an optimization pass.

[image]

alaindargelas, 2022-08-07:
Maybe because we issue this attribute for the submodule "Helper":
(* keep =  1  *)

kamilrakoczy, 2022-08-08:
We are explicite adding ``keep`` attribute in the plugin here: https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/systemverilog-plugin/UhdmAst.cc#L1735.

It was added, as sometimes modules from genscope (in opentitan) were removed by yosys (when they shouldn't be removed).

As it was added some time ago, maybe it is not needed anymore, but if it is, this needs to be investigated (as this attribute shouldn't be needed).

kamilrakoczy, 2024-07-09:
We are no longer adding `keep` attribute and sv2v `relong/split_ports.v` test is passing formal verification. Closing issue at this point.

closed 2024-07-09 by kamilrakoczy