chipsalliance/synlig #968: SystemVerilog streaming operator causes segmentation fault

jeras, 2022-08-06:
Here is the example code, I do not have a small example yet.
```
// bit reversal
function automatic logic [XLEN-1:0] bitrev (logic [XLEN-1:0] val);
`ifdef LANGUAGE_UNSUPPORTED_STREAM_OPERATOR
  for (int unsigned i=0; i<XLEN; i++)  bitrev[i] = val[XLEN-1-i];
`else
  bitrev = {<<{val}};
`endif
endfunction: bitrev
```
Could you link me to some kind of instructions on how to add a test to official regression tests, so I could prepare a pull request wit a new testcase?

The code crashes if the macro is not defined. This is just a part of the output just to give some indication at which stage the crash occured.
```
...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 9
[   NOTE] : 6
Warning: Removing unused module: \r5p_wbu from the design.
Warning: Removing unused module: \r5p_lsu from the design.
Warning: Removing unused module: \r5p_bru from the design.
Warning: Removing unused module: \r5p_gpr from the design.
Warning: Removing unused module: \r5p_core from the design.
Generating RTLIL representation for module `\r5p_alu'.
Segmentation fault (core dumped)
```

This code works correctly in Verilator, I synthesized it with Vivado, but I have not tested the build, Quartus does not support streaming operators.

alaindargelas, 2022-08-07:
This is completely supported in Surelog. Issue is in the Yosys plugin/Yosys.

jeras, 2022-08-07:
OK, I thought it was a plugin issue, since I only executed the `read_systemverilog ... -parse ...` command.

jeras, 2022-08-09:
Steps to reproduce the issue:
```
git clone https://github.com/jeras/UHDM-tests
cd UHDM-tests
../yosys-uhdm-plugin-integration/image/bin/yosys -s build_reverse.tcl
```
The output is:
```
$ ../rp32/submodules/yosys-uhdm-plugin-integration/image/bin/yosys -s build_reverse.tcl 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.16+65 (git sha1 52d8ddee0, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

-- Executing script file `build_reverse.tcl' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] test_reverse.sv:1:1: No timescale set for "test_reverse".

[INF:CP0300] Compilation...

[INF:CP0303] test_reverse.sv:1:1: Compile module "work@test_reverse".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[NTE:EL0503] test_reverse.sv:1:1: Top level module "work@test_reverse".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 1.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 1
[   NOTE] : 5
Generating RTLIL representation for module `\test_reverse'.
Segmentation fault (core dumped)
```

jeras, 2022-08-12:
I forgot to commit/push the script the last time, it is now in the repo.

rkapuscik, 2022-08-22:
Thanks for your report, we've reproduced the issue and started investigating the cause.

For the regression tests, they are located in [UHDM-Integration repository](https://github.com/chipsalliance/UHDM-integration-tests). You can just copy one of the test folders and adjust the files inside. For running the regression tests in Yosys, see the readme in that repository. It assumes that you have built yosys from the [Yosys-Systemverilog repository](https://github.com/antmicro/yosys-systemverilog/), but you can provide your own `YOSYS_BIN` (see [here](https://github.com/chipsalliance/UHDM-integration-tests/blob/8dc491bb51384418a41a7d2f9c30ac8e576f4619/Makefile#L7)).

jeras, 2022-08-24:
I created my first pull request with this simple test, please have a look.
When this goes through I will move to the other tests.

alaindargelas, 2022-08-24:
@jeras, which pull request? Please add a link to it here.

jeras, 2022-08-24:
Sorry, I was distracted by Github already linking it into the conversation above.
https://github.com/chipsalliance/UHDM-integration-tests/pull/693

mglb, 2023-05-15:
Fixed. The test https://github.com/jeras/UHDM-integration-tests/commit/87d736de151f3dd13ec360cfd1a793dba3cd2c56 passes formal verification at this moment.

closed 2023-05-15 by mglb