chipsalliance/synlig #969: Structure parameter element used as as condition in ?: operator ...

jeras, 2022-08-06:
Structure parameter element is used as as condition in ?: operator to parameterize an instance.
```
  .AW      (ISA.spec.base.E ? 4 : 5),
```
The `AW` parameter is defined as:
```
  int unsigned AW   = 5,     // can be 4 for RV32E base ISA
```
The reported error is:
```
../../../hdl/rtl/core/r5p_core.sv:333: ERROR: Identifier `\ISA' is implicitly declared and `default_nettype is set to none.
```

This is not a critical issue for me, since I have only one instance, and it was trivial to workaround.

alaindargelas, 2022-08-07:
This is modeled correctly in UHDM. Issue must be in the yosys plugin.

rkapuscik, 2022-08-22:
Thanks for reporting this, would you be able to provide a minimal complete test case as described in https://github.com/antmicro/yosys-systemverilog/issues/968#issuecomment-1221900209?