chipsalliance/synlig #970: unpacked array initialization with assignment pattern

jeras, 2022-08-06:
I tried to write an initialized memory.
```
  // register file (FPGA would initialize it to all zeros)
  logic [XLEN-1:0] gpr [0:2**AW-1] = '{default: '0};
```
The error does not provide any details indicating where is the offending code:
```
ERROR: Couldn't find ancestor for tagged pattern!
```
I removed the initialization now, since it does not affect the implementation much, initialization is usually done by SW on ASIC.

This code was only tested with Verilator.

alaindargelas, 2022-08-07:
This is also modeled correctly in UHDM.

jeras, 2022-08-08:
I created an example implementing a RISC-V GPR (register file) with W1R2 ports and asynchronous read.
On FPGA architectures supporting this it should be implemented as distributed RAM.

Steps to reproduce the issue:
```
git clone https://github.com/jeras/UHDM-tests
cd UHDM-tests
../yosys-uhdm-plugin-integration/image/bin/yosys -s build_gpr.tcl
```

The output is:
```
$ ../rp32/submodules/yosys-uhdm-plugin-integration/image/bin/yosys -s build_gpr.tcl 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.16+65 (git sha1 52d8ddee0, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

-- Executing script file `build_gpr.tcl' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] r5p_gpr.sv:19:1: No timescale set for "r5p_gpr".

[INF:CP0300] Compilation...

[INF:CP0303] r5p_gpr.sv:19:1: Compile module "work@r5p_gpr".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[INF:CP0335] r5p_gpr.sv:83:6: Compile generate block "work@r5p_gpr.gen_wb_default".

[NTE:EL0503] r5p_gpr.sv:19:1: Top level module "work@r5p_gpr".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 1
[   NOTE] : 5
ERROR: Couldn't find ancestor for tagged pattern!
```

If I define the `LANGUAGE_UNSUPPORTED_ARRAY_ASSIGNMENT_PATTERN` macro, to select the code without the memory initialization I seem to get the desired `RAM32M` distributed memory.
```
=== r5p_gpr ===

   Number of wires:                 41
   Number of wire bits:            269
   Number of public wires:          15
   Number of public wire bits:     182
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                136
     BUFG                            1
     IBUF                           53
     LUT6                            1
     MUXF7                           1
     OBUF                           64
     RAM32M                         16

   Estimated number of LCs:          1
```

rkapuscik, 2022-08-22:
Thanks for reporting this and for the test file. Wwould you be able to provide a minimal complete test case as described in https://github.com/antmicro/yosys-systemverilog/issues/968#issuecomment-1221900209?