chipsalliance/synlig #989: Parsing Verilog from Vivado

clavin-xlnx, 2022-08-11:
I am attempting to read structural Verilog generated by Vivado into Yosys.  
Here is my example Verilog input:
[gnl_4_8_7_2.0_gnl_8000_08_7_80_80.zip](https://github.com/antmicro/yosys-systemverilog/files/9311531/gnl_4_8_7_2.0_gnl_8000_08_7_80_80.zip)

Here is the output log:
[surelog.log](https://github.com/antmicro/yosys-systemverilog/files/9311509/surelog.log)

There are a number of warning of the type:
```
[WRN:EL0500] gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v:11078: Cannot find a module definition for "work@gnl_8000_08_7_80_80::FDRE".
```

My goal is simply to read in the Verilog and write out an equivalent `.json` file suitable for place and route in `nextpnr`.

tgorochowik, 2022-08-12:
Thank you for the report!

To get rid of the warnings you need to explicitly read the xilinx blackboxes definitions from yosys (https://github.com/YosysHQ/yosys/blob/master/techlibs/xilinx/cells_sim.v) otherwise it won't find module definitions for FDRE, OBUF, IBUF etc.

So like this:
```
plugin -i systemverilog
read_systemverilog yosys/blob/master/techlibs/xilinx/cells_sim.v gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v
```

Unfortunately there are some other issues too, something triggers an assert in RTLIL for your design. We will take a look at this and come back to you.

alaindargelas, 2022-08-17:
Please note the proper Surelog command to load this testcase would be: 
surelog -parse -elabuhdm gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v -v cells_sim.v

Using -v for cells_sim.v will prevent the multiple top level modules due to unused cells.

plugin -i systemverilog
read_systemverilog gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v -v yosys/blob/master/techlibs/xilinx/cells_sim.v

rkapuscik, 2022-08-18:
Thanks for your patience! After properly adding the cells the design errors out with
```
Generating RTLIL representation for module `\gnl_8000_08_7_80_80'.
ERROR: Assert `!wire->name.empty()' failed in kernel/rtlil.cc:2017.
```
It seems that some information is missing in the UHDM (see https://github.com/chipsalliance/Surelog/issues/3156), so first fix needs to happen in Surelog, then we will revisit it in the plugin.

alaindargelas, 2022-08-22:
@rkapuscik, https://github.com/chipsalliance/Surelog/issues/3156 is fixed.

alaindargelas, 2022-09-01:
@rkapuscik I put a note in the case, you have all the information you need. You need to get it from a few places.

alaindargelas, 2022-09-10:
@rkapuscik that case is fixed in Surelog, per your spec.

closed 2023-03-15 by tgorochowik