chipsalliance/synlig #994: SystemVerilog interface related broad issue

jeras, 2022-08-12:
The `tcb_gpio.sv` RTL is a simple GPIO peripheral with `tcb_if.sv` SystemVerilog interface as the system bus.
`tcb_gpio_wrap.sv` is just a wrapper providing simple signals for ports.

The design was tested in Vivado and consumed the expected 96 FFs.

The issues are:
1. `tcb_gpio` module is apparently removed.
2. Please explain warnings like `Warning: Identifier `\bus.vld' is implicitly declared.`
3. `Warning: wire '\bus.rdt' is assigned in a block at tcb_gpio.sv:56.0-56.0.` is incorrect, since there are no wires in the design, all signals are of type `logic`.
4. `ERROR: Failed to resolve identifier \bus.adr for width detection!`

A few comments:
- The signal `trn = vld & rdy` is defined as RTL inside the interface.
- There is some parameter validation code which is just experimental. I am now to it, I do not expect it to work.

To reproduce this issue do:
```
git clone https://github.com/jeras/UHDM-tests
cd UHDM-tests
../yosys-uhdm-plugin-integration/image/bin/yosys -s build_gpio.tcl
```

This is my output:
```
$ ../rp32/submodules/yosys-uhdm-plugin-integration/image/bin/yosys -s build_gpio.tcl 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.16+65 (git sha1 52d8ddee0, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

-- Executing script file `build_gpio.tcl' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] tcb_if.sv:19:1: No timescale set for "tcb_if".

[WRN:PA0205] tcb_gpio.sv:23:1: No timescale set for "tcb_gpio".

[WRN:PA0205] tcb_gpio_wrap.sv:19:1: No timescale set for "tcb_gpio_wrap".

[INF:CP0300] Compilation...

[INF:CP0303] tcb_gpio.sv:23:1: Compile module "work@tcb_gpio".

[INF:CP0303] tcb_gpio_wrap.sv:19:1: Compile module "work@tcb_gpio_wrap".

[INF:CP0304] tcb_if.sv:19:1: Compile interface "work@tcb_if".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[INF:CP0335] tcb_if.sv:57:8: Compile generate block "work@tcb_gpio_wrap.bus.gen_dly".

[INF:CP0335] tcb_if.sv:58:5: Compile generate block "work@tcb_gpio_wrap.bus.gen_dly.gen_rsp".

[NTE:EL0503] tcb_gpio_wrap.sv:19:1: Top level module "work@tcb_gpio_wrap".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 2.

[NTE:EL0510] Nb instances: 2.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 3
[   NOTE] : 5
Warning: Removing unused module: \tcb_gpio from the design.
Generating RTLIL representation for module `\tcb_gpio_wrap'.
tcb_gpio_wrap.sv:51: Warning: Identifier `\bus.vld' is implicitly declared.
tcb_gpio_wrap.sv:52: Warning: Identifier `\bus.wen' is implicitly declared.
tcb_gpio_wrap.sv:53: Warning: Identifier `\bus.adr' is implicitly declared.
tcb_gpio_wrap.sv:54: Warning: Identifier `\bus.ben' is implicitly declared.
tcb_gpio_wrap.sv:55: Warning: Identifier `\bus.wdt' is implicitly declared.
tcb_gpio_wrap.sv:57: Warning: Identifier `\bus.rdt' is implicitly declared.
tcb_gpio_wrap.sv:58: Warning: Identifier `\bus.err' is implicitly declared.
tcb_gpio_wrap.sv:59: Warning: Identifier `\bus.rdy' is implicitly declared.
Generating RTLIL representation for module `$paramod\tcb_gpio\GW=32'd32'.
Warning: wire '\bus.rdt' is assigned in a block at tcb_gpio.sv:56.0-56.0.
tcb_gpio.sv:59: ERROR: Failed to resolve identifier \bus.adr for width detection!
```

alaindargelas, 2022-08-22:
I'll check the Surelog part shortly.

rkapuscik, 2022-08-22:
Thanks for your report! I'll try to address some of your questions.
`tcb_gpio` is a parameterized module, you can see in the output that the parameterized version is preserved: 
```
Generating RTLIL representation for module `$paramod\tcb_gpio\GW=32'd32'.
```
`\tcb_gpio` is partial version which is expected to be removed.
Implicitly declared identifiers suggest that the interface definition wasn't parsed properly. This would also explain error with width detection. My first guess would be to check whether parameters are handled correctly in the plugin code.

Thanks for providing the source files, they are great for reproducing the problem, but for fixing it it would be great to have a minimal example of the failing code (like those in [UHDM integration repository](https://github.com/chipsalliance/UHDM-integration-tests/tree/master/tests)). Would you be able to reduce your test case?

jeras, 2022-08-22:
I will have a look at other regression tests and do something similar.
I reported many bugs and provided regression test cases for Verilator, so I have some experience with this.
I will just take me some time.

alaindargelas, 2022-08-23:
@rkapuscik, I created the following testcase to check the UHDM tree, and all seems correct:

```

interface tcb_if #(
  // bus widths
  int unsigned AW = 32,    // address width
  int unsigned DW = 32,    // data    width
  int unsigned BW = DW/8,  // byte e. width
  // response delay
  int unsigned DLY = 1
)(
  // system signals
  input  logic clk,  // clock
  input  logic rst   // reset
);

  // system bus
  logic          vld;  // valid
  
  // transfer (valid and ready at the same time)
  assign trn = vld & rdy;
  // idle (either not valid or ending a cycle with a transfer)
  assign idl = ~vld | trn;
  // response (DLY clock periods after transfer)
  generate
  if (DLY == 0) begin: gen_rsp
    assign rsp = trn;
  end: gen_rsp
  // response delay queue
  else begin: gen_dly
    if (DLY == 1) begin: gen_rsp
      always @(posedge clk, posedge rst)
      if (rst)  rsp <= 1'b0;
      else      rsp <= trn & ~wen;
    end: gen_rsp
  end: gen_dly
  endgenerate
 
  // manager
  modport  man (
   
    output vld
   
  );

  // monitor
  modport  mon (
    
    input  vld
    
  );

  // subordinate
  modport  sub (
   
    input  vld
   
  );

endinterface: tcb_if

module tcb_gpio #(
  // GPIO parameters
  int unsigned GW = 32  // GPIO width
)(
  // system bus interface
  tcb_if.sub bus
);

[... 1159 lines cut ...]
      \_cont_assign: , id:238, line:99:8, endln:99:25
      |vpiName:bus_vld
      |vpiFullName:work@tcb_gpio_wrap.bus_vld
      |vpiActual:
      \_logic_net: (bus_vld), id:210
    |vpiLhs:
    \_hier_path: (bus.vld), id:239, line:99:8, endln:99:15
      |vpiParent:
      \_cont_assign: , id:238, line:99:8, endln:99:25
      |vpiName:bus.vld
      |vpiActual:
      \_ref_obj: (bus), id:240
        |vpiParent:
        \_hier_path: (bus.vld), id:239, line:99:8, endln:99:15
        |vpiName:bus
        |vpiActual:
        \_interface: work@tcb_if (work@tcb_gpio_wrap.bus), id:211, file:tests/ScratchPad.sv, line:97:1, endln:97:59
      |vpiActual:
      \_ref_obj: (vld), id:241
        |vpiParent:
        \_hier_path: (bus.vld), id:239, line:99:8, endln:99:15
        |vpiName:vld
        |vpiActual:
        \_io_decl: (vld), id:221, line:41:12, endln:41:15
```