chipsalliance/synlig #1041: problems with -defer

dpetrisko, 2022-08-31:
Hi, I'm having trouble using surelog+yosys. You can retrieve my test SV here if you wish to reproduce: https://github.com/bespoke-silicon-group/basejump_stl

I am using the default submodules except for https://github.com/chipsalliance/yosys-f4pga-plugins/pulls/377 to fix https://github.com/chipsalliance/yosys-f4pga-plugins/issues/380

Here is my full yosys script:
```
# Get systemverilog plugin
plugin -i systemverilog
# Set up include files
systemverilog_defaults -add -Ibasejump_stl/bsg_misc
systemverilog_defaults -add -Ibasejump_stl/bsg_cache
# Read files one at a time
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_pkg.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_mux.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_mux_segmented.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_circular_ptr.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff_en.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff_en_bypass.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_counter_clear_up.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_decode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_priority_encode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_encode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_decode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_backup.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_mux_bitwise.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_priority_encode_one_hot_out.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_encode_one_hot.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_scan.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_expand_bitmask.v
read_systemverilog -defer basejump_stl/bsg_fsb/bsg_fsb_node_trace_replay.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1r1w_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1r1w.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_two_fifo.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small_unhardened.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_tracker.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_decode.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_dma.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_miss.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_sbuf.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache.v
# Finish reading files, elaborate the design
read_systemverilog -link
```

but in fact, it doesn't seem to be getting very far at all. It seems to be erroring out on the first file read despite the -defer flag

```
yosys> plugin -i systemverilog

yosys> systemverilog_defaults -add -Ibasejump_stl/bsg_misc

yosys> systemverilog_defaults -add -Ibasejump_stl/bsg_cache

yosys> read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_pkg.v
1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 3
[   NOTE] : 0
ERROR: Error when parsing design. Aborting!
```

My slpp_all/surelog.log is:

```
********************************************
*  SURELOG SystemVerilog  Compiler/Linter  *
********************************************

Copyright (c) 2017-2022 Alain Dargelas,
http://www.apache.org/licenses/LICENSE-2.0

VERSION: 1.36
BUILT  : Aug 31 2022
DATE   : 2022-08-31.16:21:28
COMMAND: basejump_stl/bsg_cache/bsg_cache_pkg.v -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache -PYOSYS=1

[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 3
[   NOTE] : 0
```

which seems to be the right command, it just seems to be happening too early.

Thanks for the help!

alaindargelas, 2022-09-01:
I created the following bash script outside of Yosys (which basically does the same thing):

```
surelog -init
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_pkg.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_mux.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_mux_segmented.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_circular_ptr.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_dff.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_dff_en.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_dff_en_bypass.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_counter_clear_up.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_decode.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_priority_encode.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_lru_pseudo_tree_encode.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_lru_pseudo_tree_decode.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_lru_pseudo_tree_backup.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_mux_bitwise.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_priority_encode_one_hot_out.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_encode_one_hot.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_scan.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_misc/bsg_expand_bitmask.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_fsb/bsg_fsb_node_trace_replay.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1r1w_synth.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1r1w.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync_synth.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte_synth.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_dataflow/bsg_two_fifo.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small_unhardened.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_dataflow/bsg_fifo_tracker.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_decode.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_dma.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_miss.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_sbuf.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v
surelog -sepcomp -Ibasejump_stl/bsg_misc -Ibasejump_stl/bsg_cache basejump_stl/bsg_cache/bsg_cache.v
surelog -link
```

And I got a real syntax error that is somehow masked by the Yosys plugin code:

....
[SNT:PA0207] basejump_stl/bsg_misc/bsg_circular_ptr.v:60:5: Syntax error: extraneous input 'else' expecting {';', 'default', 'module', 'endmodule', 'extern', 'macromodule', 'interface', 'program', 'virtual', 'class', 'timeunit', 'timeprecision', 'checker', 'type', 'clocking', 'defparam', 'bind', 'const', 'function', 'static', 'constraint', 'if', 'automatic', 'localparam', 'parameter', 'specparam', 'import', 'genvar', 'typedef', 'enum', 'struct', 'union', 'string', 'chandle', 'event', '[', 'byte', 'shortint', 'int', 'longint', 'integer', 'time', 'bit', 'logic', 'reg', 'shortreal', 'real', 'realtime', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0', 'tri1', 'wire', 'uwire', 'wand', 'wor', 'trireg', 'signed', 'unsigned', 'interconnect', 'var', '$', 'export', DOLLAR_UNIT, '(*', 'assert', 'property', 'assume', 'cover', 'not', 'or', 'and', 'sequence', 'covergroup', 'pulldown', 'pullup', 'cmos', 'rcmos', 'bufif0', 'bufif1', 'notif0', 'notif1', 'nmos', 'pmos', 'rnmos', 'rpmos', 'nand', 'nor', 'xor', 'xnor', 'buf', 'tranif0', 'tranif1', 'rtranif1', 'rtranif0', 'tran', 'rtran', 'generate', 'case', 'for', 'global', 'initial', 'assign', 'alias', 'always', 'always_comb', 'always_latch', 'always_ff', 'restrict', 'let', 'this', 'randomize', 'final', 'task', 'specify', 'sample', '=', 'nettype', Escaped_identifier, Simple_identifier, '`pragma', SURELOG_MACRO_NOT_DEFINED},
     else
     ^-- ./slpp_all/work/bsg_misc_10443116261081232805/bsg_circular_ptr.v:60:5:.

dpetrisko, 2022-09-01:
Hm, it looks likes the macro `BSG_IS_POW2 is not being picked up on L40 [here](https://github.com/bespoke-silicon-group/basejump_stl/blob/46ea2b163a96e14c1d255a9edcad4f7163727f99/bsg_misc/bsg_circular_ptr.v#L49). It is defined in bsg_defines.v [here](https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_defines.v) which is included at the top [here](https://github.com/bespoke-silicon-group/basejump_stl/blob/46ea2b163a96e14c1d255a9edcad4f7163727f99/bsg_misc/bsg_circular_ptr.v#L1)

Perhaps when we defer compilation we're not defining macros needed to build the AST?

alaindargelas, 2022-09-01:
These files are .v file yet they contain some SystemVerilog constructs,
they should be .sv
To override, add -sverilog on the command line of the offending
file basejump_stl/bsg_misc/bsg_circular_ptr.v

alaindargelas, 2022-09-01:
@koluckirafal why Syntax errors are filtered out by the plugin ? That makes debugging harder.

alaindargelas, 2022-09-01:
Your script should be:

```
# Get systemverilog plugin
plugin -i systemverilog
# Set up include files
systemverilog_defaults -add -Ibasejump_stl/bsg_misc
systemverilog_defaults -add -Ibasejump_stl/bsg_cache
# Read files one at a time
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_pkg.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_mux.v
read_systemverilog -defer -sverilog basejump_stl/bsg_misc/bsg_mux_segmented.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_circular_ptr.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff_en.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_dff_en_bypass.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_counter_clear_up.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_decode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_priority_encode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_encode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_decode.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_lru_pseudo_tree_backup.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_mux_bitwise.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_priority_encode_one_hot_out.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_encode_one_hot.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_scan.v
read_systemverilog -defer basejump_stl/bsg_misc/bsg_expand_bitmask.v
read_systemverilog -defer basejump_stl/bsg_fsb/bsg_fsb_node_trace_replay.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1r1w_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1r1w.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte_synth.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit.v
read_systemverilog -defer basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_byte.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_two_fifo.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_1r1w_small_unhardened.v
read_systemverilog -defer basejump_stl/bsg_dataflow/bsg_fifo_tracker.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_decode.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_dma.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_miss.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_sbuf.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v
read_systemverilog -defer basejump_stl/bsg_cache/bsg_cache.v
# Finish reading files, elaborate the design
read_systemverilog -link

```

dpetrisko, 2022-09-01:
Thanks Alain I'll give that a shot!

dpetrisko, 2022-09-01:
No luck, same error with a reduced testcase:

script 1:
```
# Get systemverilog plugin
plugin -i systemverilog
# Set up include files
systemverilog_defaults -add -sverilog
systemverilog_defaults -add -Ibasejump_stl/bsg_misc
# Read files one at a time
read_systemverilog -defer -pvalue+slots_p=4 -pvalue+max_add_p=1 basejump_stl/bsg_misc/bsg_circular_ptr.v
# Finish reading files, elaborate the design
read_systemverilog -link
```
script 2:
```
# Get systemverilog plugin
plugin -i systemverilog
# Set up include files
systemverilog_defaults -add -Ibasejump_stl/bsg_misc
# Read files one at a time
read_systemverilog -defer -sverilog -pvalue+slots_p=4 -pvalue+max_add_p=1 basejump_stl/bsg_misc/bsg_circular_ptr.v
# Finish reading files, elaborate the design
read_systemverilog -link
```

dpetrisko, 2022-09-01:
Notably this minimal surelog example fails for me as well, though with a different error

script 3:
```
surelog -init
surelog -sepcomp -Ibasejump_stl/bsg_misc -pvalue+slots_p=4 -pvalue+max_add_p=1 -sv basejump_stl/bsg_misc/bsg_circular_ptr.v
surelog -link
```

Output:
```
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 3
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[ERR:PA0213] Internal error: CACHE OUT OF BOUND.

fail.script: line 3: 33331 Segmentation fault      surelog -link
```

alaindargelas, 2022-09-02:
@dpetrisko looks like you have an older build of Surelog. Can you checkout Surelog from main. This test pass for me:

 ./Surelog/build/bin/surelog -sepcomp -Ibasejump_stl/bsg_misc -pvalue+slots_p=4 -pvalue+max_add_p=1 -sv basejump_stl/bsg_misc/bsg_circular_ptr.v
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: No timescale set for "bsg_circular_ptr".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_circular_ptr__abstract".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 5
[   NOTE] : 0
alain@alain-xps:~$ ./Surelog/build/bin/surelog -link
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: No timescale set for "bsg_circular_ptr".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_circular_ptr__abstract".

[INF:CP0300] Compilation...

[INF:CP0303] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: Compile module "work@bsg_circular_ptr".

[INF:CP0303] basejump_stl/bsg_misc/bsg_defines.v:29:5: Compile module "work@bsg_circular_ptr__abstract".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[NTE:CP0309] basejump_stl/bsg_misc/bsg_circular_ptr.v:19:33: Implicit port type (wire) for "o",
there are 1 more instances of this message.

[INF:EL0526] Design Elaboration...

[NTE:EL0503] basejump_stl/bsg_misc/bsg_defines.v:29:5: Top level module "work@bsg_circular_ptr__abstract".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 1.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: ./slpp_all/surelog.uhdm ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 2
[   NOTE] : 6

alaindargelas, 2022-09-02:
Note maybe the plugin still points to an older Surelog too?

alaindargelas, 2022-09-02:
And the following script:

```
# Get systemverilog plugin
plugin -i systemverilog
# Read files one at a time
read_systemverilog -defer -sverilog  -Ibasejump_stl/bsg_misc -pvalue+slots_p=4 -pvalue+max_add_p=1 basejump_stl/bsg_misc/bsg_circular_ptr.v
# Finish reading files, elaborate the design
read_systemverilog -link -d uhdm
```

Produces the following output:

```
./yosys-uhdm-plugin-integration/image/bin/yosys -s yosys.ys 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.16+65 (git sha1 52d8ddee0, gcc 11.2.0-7ubuntu2 -fPIC -Os)

-- Executing script file `yosys.ys' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: No timescale set for "bsg_circular_ptr".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_circular_ptr__abstract".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 5
[   NOTE] : 0

2. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: No timescale set for "bsg_circular_ptr".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_circular_ptr__abstract".

[INF:CP0300] Compilation...

[INF:CP0303] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: Compile module "work@bsg_circular_ptr".

[... 315 lines cut ...]
  |vpiParent:
  \_design: (work@bsg_circular_ptr__abstract)
  |vpiFullName:work@bsg_circular_ptr__abstract
  |vpiDefName:work@bsg_circular_ptr__abstract
|uhdmtopModules:
\_module: work@bsg_circular_ptr__abstract (work@bsg_circular_ptr__abstract), file:basejump_stl/bsg_misc/bsg_defines.v, line:29:5, endln:29:87
  |vpiName:work@bsg_circular_ptr__abstract
  |vpiDefName:work@bsg_circular_ptr__abstract
  |vpiTop:1
  |vpiTopModule:1
===================
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 2
[   NOTE] : 6
Warning: Removing unused module: \bsg_circular_ptr from the design.
Generating RTLIL representation for module `\bsg_circular_ptr__abstract'.

Warnings: 1 unique messages, 1 total
End of script. Logfile hash: e954377ab5, CPU: user 0.02s system 0.00s, MEM: 24.35 MB peak
Yosys 0.16+65 (git sha1 52d8ddee0, gcc 11.2.0-7ubuntu2 -fPIC -Os)
Time spent: 79% 4x read_systemverilog (0 sec), 20% 1x plugin (0 sec)

```

dpetrisko, 2022-09-02:
Hi Alain,

Latest surelog gives the same error for me on your script above:

```
commit ab317007e7313ecd8643a65e35e12c88c058d6f0 (HEAD, origin/master, origin/HEAD, master)
Merge: 922672d47 6c58db95a
Author: alaindargelas <63669492+alaindargelas@users.noreply.github.com>
Date:   Wed Aug 31 13:09:02 2022 -0700

    Merge pull request #3197 from hs-apotell/platform-logs

    Add support for platform specific logs for regression tests
```

```
[ERR:PA0213] Internal error: CACHE OUT OF BOUND.
```

I guess this could be an environment problem?

dpetrisko, 2022-09-02:
Compiling with GCC 11 vs GCC 9 seems to have fixed it... will update if I find anything else

alaindargelas, 2022-09-02:
Alright!

alaindargelas, 2022-09-02:
I'll add a note to the readme of Surelog

closed 2022-09-02 by dpetrisko