chipsalliance/synlig #1064: ERROR: Failed to detect width for parameter

dpetrisko, 2022-09-07:
Hi, trying to track down an issue, I believe it's in the plugin but it could also be in surelog.

I am using the default submodules except for https://github.com/chipsalliance/yosys-f4pga-plugins/pulls/377 to fix https://github.com/chipsalliance/yosys-f4pga-plugins/issues/380 and https://github.com/chipsalliance/Surelog/pull/3208 to fix https://github.com/chipsalliance/Surelog/issues/3203.

To get the code:

```
git clone git@github.com:black-parrot/black-parrot
git submodule update --init --recursive
```

Here is a script which can successfully parse BlackParrot:
[surelog.txt](https://github.com/chipsalliance/yosys-f4pga-plugins/files/9507818/surelog.txt) 

```bash surelog.txt```

Here is a script which fails to link BlackParrot.

[yosys.txt](https://github.com/chipsalliance/yosys-f4pga-plugins/files/9507819/yosys.txt)

```yosys -s blackparrot.txt```

Here is the tail of the output:
```
===================
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 346
[   NOTE] : 258
Warning: black-parrot/bp_be/src/v/bp_be_calculator/bp_be_calculator_top.sv:400: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/bp_be/src/v/bp_be_calculator/bp_be_calculator_top.sv:458: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:1018: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/bp_be/src/v/bp_be_checker/bp_be_detector.sv:142: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/bp_fe/src/v/bp_fe_icache.sv:502: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_cache/bsg_cache.v:895: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_cache/bsg_cache_miss.v:332: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_cache/bsg_cache_miss.v:353: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_cache/bsg_cache_miss.v:426: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_misc/bsg_expand_bitmask.v:34: Post-incrementation operations are handled as pre-incrementation.
Warning: black-parrot/external/basejump_stl/bsg_dataflow/bsg_serial_in_parallel_out.v:87: Post-incrementation operations are handled as pre-incrementation.
black-parrot/bp_common/src/include/bp_common_aviary_defines.svh:63: ERROR: Failed to detect width for parameter \proc_param_lp!
```

We can see 0 FATAL, SYNTAX or ERROR during parsing, but an ERROR during the linking step.  bp_common_aviary_defines.svh is basically a macro which assigns localparams from a large localparam array. Oddly, the line in question is about halfway through the file and there doesn't appear to be anything special about it.  I would expect the strategy to either completely succeed or completely fail.  What could be going on here? Thanks!

alaindargelas, 2022-09-07:
@rkapuscik it is an error coming from the plugin.

dpetrisko, 2022-09-19:
Hi, is there any additional information I could give to help debug this?

alaindargelas, 2022-09-19:
@rkapuscik was out sick, should be back at it soon

mglb, 2022-09-22:
@dpetrisko file lists on the surelog script and the yosys script differ slightly. Files missing in the yosys script:
```
black-parrot/external/basejump_stl/bsg_misc/bsg_array_concentrate_static.v
black-parrot/external/basejump_stl/bsg_async/bsg_launch_sync_sync.v
black-parrot/external/basejump_stl/bsg_async/bsg_sync_sync.v
```
and there's extra `black-parrot/external/basejump_stl/bsg_misc/bsg_defines.v`. I've attached a bash script that can run either yosys or surelog (with `bash process_design.txt yosys` or `bash process_design.txt surelog`).
[process_design.txt](https://github.com/antmicro/yosys-systemverilog/files/9627045/process_design.txt)

Anyway, even with updated file list, I've reproduced your problem.
Can you run `yosys -s blackparrot.txt 2>&1 | tee yosys.log` and check whether there are `FILE INDEX MISMATCH` lines?
I got lines like these, from both scripts (yosys and surelog):
```
[INF:CP0303] black-parrot/external/HardFloat/source/addRecFN.v:204:1: Compile module "work@addRecFN".
[INF:CP0303] black-parrot/external/HardFloat/source/addRecFN.v:43:1: Compile module "work@addRecFNToRaw".
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_calculator_top.sv:18:1: Compile module "work@bp_be_calculator_top".
FILE INDEX MISMATCH
[ERR:CP0336] Internal out of bound error.
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_checker/bp_be_cmd_queue.sv:5:1: Compile module "work@bp_be_cmd_queue".
FILE INDEX MISMATCH
[ERR:CP0336] Internal out of bound error.
FILE INDEX MISMATCH
FILE INDEX MISMATCH
FILE INDEX MISMATCH
FILE INDEX MISMATCH
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:5:1: Compile module "work@bp_be_csr".
FILE INDEX MISMATCH
[ERR:CP0336] Internal out of bound error.
FILE INDEX MISMATCH
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:91:1: Compile module "work@bp_be_dcache".
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache_decoder.sv:7:1: Compile module "work@bp_be_dcache_decoder".
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache_wbuf.sv:12:1: Compile module "work@bp_be_dcache_wbuf".
[INF:CP0303] black-parrot/bp_be/src/v/bp_be_checker/bp_be_detector.sv:17:1: Compile module "work@bp_be_detector".
```
`FILE INDEX MISMATCH` is printed to stderr from Surelog (https://github.com/chipsalliance/Surelog/blob/b4ad3fc02ef02ad162f34ac7d18c397e5fd6e9d9/src/Design/FileContent.cpp#L689). For some reason the error message added in a line above is not printed.

dpetrisko, 2022-09-22:
Hi yes, I also see these FILE INDEX MISMATCH lines

dpetrisko, 2022-09-22:
(In both yosys and surelog)

alaindargelas, 2022-09-23:
The "FILE INDEX MISMATCH" is not the problem at all.
This is a simple debug message about file and line number. Not your concern. I'll ask the guy who introduced this new message in Surelog. It is irrelevant to the problem at hand is this:
black-parrot/bp_common/src/include/bp_common_aviary_defines.svh:63: ERROR: Failed to detect width for parameter \proc_param_lp!

please make sure your build of Surelog is uptodate.
Also remove the cache in your current directory where you build the design:   rm -rf slpp_unit slpp_all

alaindargelas, 2022-09-23:
That is what I see in Surelog:
```
alain@alain-xps:~$ rm -rf slpp_all/
alain@alain-xps:~$ rm -rf slpp_unit/
alain@alain-xps:~$ source ./black.sh
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_cache/bsg_cache_pkg.v:9:1: No timescale set for "bsg_cache_pkg".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 4
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_mux.v:2:1: No timescale set for "bsg_mux".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_mux__abstract".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 5
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_mux_segmented.v:9:1: No timescale set for "bsg_mux_segmented".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_defines.v:29:5: No timescale set for "bsg_mux_segmented__abstract".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 5
[   NOTE] : 0
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_misc/bsg_circular_ptr.v:11:1: No timescale set for "bsg_circular_ptr".
[... 1994 lines cut ...]

[NTE:EL0503] basejump_stl/bsg_misc/bsg_defines.v:29:5: Top level module "work@bsg_mem_1r1w_synth__abstract".

[NTE:EL0503] basejump_stl/bsg_misc/bsg_defines.v:29:5: Top level module "work@bsg_mem_1rw_sync_synth__abstract".

[NTE:EL0504] Multiple top level modules in design.

[NTE:EL0508] Nb Top level modules: 34.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 34.

[NTE:EL0511] Nb leaf instances: 33.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: ./slpp_all/surelog.uhdm ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 70
[   NOTE] : 52
```

alaindargelas, 2022-09-23:
Using the proces_design.sh surelog (Which has a very different file list from the original one from Dan) I get:

```
...

[NTE:EL0504] Multiple top level modules in design.

[NTE:EL0531] black-parrot/external/basejump_stl/bsg_misc/bsg_array_concentrate_static.v:8:14: Negative value in instance "work@bsg_array_concentrate_static"
             text:   (input    [dense_els_lp-1:0][width_p-1:0] i
             value: INT:-1.

[NTE:EL0531] black-parrot/external/HardFloat/source/HardFloat_primitives.v:63:17: Negative value in instance "work@addRecFN.addRecFNToRaw.lowMask_far_roundExtraMask"
             text:         input [(inWidth - 1):0] in,
             value: INT:-1.

[NTE:EL0508] Nb Top level modules: 146.

[NTE:EL0509] Max instance depth: 16.

[NTE:EL0510] Nb instances: 2003.

[NTE:EL0511] Nb leaf instances: 494.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: ./slpp_all/surelog.uhdm ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 389
[   NOTE] : 269
```

alaindargelas, 2022-09-23:
I'm not sure why we have so many top level modules, that does not sound right. Those file lists are not representative of a "design" more like a library.

dpetrisko, 2022-09-23:
The extra top levels are included from BaseJump STL library files as "abstract modules". Basically each module is instantiated in an if (0) block in its source file, in order to avoid elaborating with default parameters which may be invalid. 

 You can see the discussion here: https://github.com/bespoke-silicon-group/basejump_stl/blob/8ccd91a5cfb252e096a70640d3dd1bb9e0e1fa46/bsg_misc/bsg_defines.v#L12. 

There's only 1 BlackParrot-specific toplevel

dpetrisko, 2022-09-23:
Ignoring the FILE_INDEX_MISMATCH, I don't see any other errors from Surelog/Yosys in the logs except for the ERROR: Failed to detect width for parameter \proc_param_lp!

alaindargelas, 2022-09-23:
@dpetrisko that is correct. @mglb please help fix the issue in the plugin code.

mglb, 2022-09-24:
I'm working on it.
So far I've found out that the plugin passes nodes with plugin's custom internal type (AST_DOT) to yosys, which is unable to interpret them.
Now I need to determine why and where exactly this happens.

alaindargelas, 2022-09-25:
The bogus error message FILE_INDEX_MISMATCH
is fixed with https://github.com/chipsalliance/Surelog/pull/3242

mglb, 2022-10-04:
Turns out the problem is (most probably) caused by another issue. The design uses union as a struct member, which is not supported by the plugin.
We're working on implementation of unions in structs, which should fix the original problem.

dpetrisko, 2022-10-04:
Thanks for the update! We fairly heavily rely on struct of unions as a construct, so if that's on the roadmap we'll wait rather than re-writing the RTL

mglb, 2022-10-18:
PR adding support for unions in structs: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/389
New enum-related issue to be fixed: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/390

I've also found the main (and hopefully final) problem. Turns out the reported symbol is most probably OK. It is just evaluated too early, and this early evaluation is caused by potentially unrelated code.
This is a bug in the plugin.

## Bug details:

Consider the following example:

```systemverilog
// --- test.sv: ---

module test(input clk_i, input reset_i);
 typedef enum bit[1:0] { test_e_0 = 0, test_e_1 = 1 } test_e;
 parameter int test_p_int_vec [1:0] = '{ 4, 8 };

 // localparam test_e test_lp_index = 1; // (A1) OK
 localparam test_e test_lp_index = test_e_1; // (A2) "ERROR: Failed to detect width for parameter \test_lp_val!" when used with B2.

 localparam test_lp_val = test_p_int_vec[test_lp_index];

 typedef struct packed {
   logic [test_lp_val-1:0]   test_s_member_0;
   logic [test_lp_val:0]     test_s_member_1;
 } test_typedef_s;

 test_helper test_helper_instance();
endmodule
```
```systemverilog
// --- test_helper.sv ---
module test_helper;

 typedef struct packed {
   integer unsigned helper_s_member;
 }  helper_typedef_s;

 // localparam integer unsigned helper_lp = 0; // (B1) ALWAYS OK
 localparam helper_typedef_s helper_lp = '{ helper_s_member: 0 }; // (B2) causes error when used with A2

endmodule
```

### Differences between assignment of integer iteral (A1) vs enum value (A2) to `test_lp_index`

- Integer literal is a node of type `constant` -> nothing to simplify.
- Enum value is a node of type `ref_obj`, which points to `enum_const` which contains `constant` -> needs simplification to become `constant`.

This is important below.

### `typedef struct` (B2) vs `int` (B1) as a param type in helper module

Actions performed inside the plugin:

- processing `test` module as *non-instance*
  - visiting various subnodes and just creating objects representing them. When instantiation of another module is visited, it needs to be processed as *instance* to create an object for it.
    - processing `test_helper` module as *instance* (`test` module instantiates it)
      - visiting all parameter nodes. If parameter's value node's type is not `constant` (B1 is `constant`, B2 is not) it is *simplified*:
        - for every typedef, parameter, and enum present in *current top node*:
          - store in a map link between object name and the object itself
          - when the object is a typedef, additionally call `simplify` (function provided by yosys) on it.
            - `test_typedef_s` is a `typedef struct` in `test` module. Value (`test_lp_val`) used in ranges of its members was not yet evaluated nor simplified (i.e. converted to `constant` node). This is why A2 causes problems here and A1 does not.

mglb, 2022-11-07:
FYI: support for unions in struct is merged: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/389

mglb, 2022-11-16:
The reported problem is technically fixed! https://github.com/chipsalliance/yosys-f4pga-plugins/pull/399
However, there are still other issues that have to be fixed in order to make blackparrot work.
The next one: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/400

mglb, 2022-11-17:
Lets keep this as a "summary" issue. I've changed the title to reflect this.

alaindargelas, 2022-12-06:
@dpetrisko I pulled the latest from black-parrot master and the file lists attached to this case don't work anymore, can you update the file list?

[FTL:CM0008] Verilog File "/home/alain/black-parrot/external/basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v" does not exist.

dpetrisko, 2022-12-06:
Hi Alain,

Attached: [flist.txt](https://github.com/antmicro/yosys-systemverilog/files/10171329/flist.txt)

Here are the tags for what I'm using (master):
black-parrot: 6ce6523d6b3ab91cdb355d53c419f1b9ee1e1e81
And the submodule tags, in case those aren't working
black-parrot/external/basejump_stl: 6a43980bc6c1deccb67f554b6f04df812e5f198a
black-parrot/external/HardFloat: f75f8f573230a4d2627d2c404599998d6d6ae475

alaindargelas, 2022-12-07:
Fixed, -sverilog on the command line fixed it (Treat .v as systemverilog files).

alaindargelas, 2022-12-07:
Elaboration works fine in Surelog:

....

[NTE:EL0508] Nb Top level modules: 146.

[NTE:EL0509] Max instance depth: 16.

[NTE:EL0510] Nb instances: 2003.

[NTE:EL0511] Nb leaf instances: 494.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: /home/alain/slpp_all/surelog.uhdm ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 450
[   NOTE] : 270

dpetrisko, 2022-12-07:
Hi Alain, I'm using the script that @mglb provided above:

[process_design.txt](https://github.com/antmicro/yosys-systemverilog/files/10179992/process_design.txt)

```
bash process_design.txt surelog
or
bash process_design.txt yosys
```

to run

dpetrisko, 2022-12-08:
Resolved it (although I certainly could have messed up)

diff from e23ff6db487da9ceea576c53ac33853566c3a84e:

[diff.txt](https://github.com/antmicro/yosys-systemverilog/files/10182911/diff.txt)

mglb, 2022-12-08:
@dpetrisko I can confirm the segfault on my end. I did look at it briefly; it seems to be caused by invalid handling of parameters without default value which are used in port's range. This is the plugin issue.

mglb, 2022-12-20:
Issue for the segfault: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/424
I'll work on this more this week.

dpetrisko, 2022-12-21:
Thank you for the update! Let me know if there's anything I can help with the provided tests!

mglb, 2022-12-24:
Turns out https://github.com/chipsalliance/yosys-f4pga-plugins/issues/424 is not an issue here.

The segfault comes from lack of support for streaming operators in the plugin: https://github.com/antmicro/yosys-systemverilog/issues/968
This will be the next thing to work on.

alaindargelas, 2023-01-21:
@dpetrisko with the latest source of black-parrot, I get the following error with the compilation script:
[FTL:CM0008] Verilog File "/home/alain/black-parrot/external/basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v" does not exist.

[  FATAL] : 1
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0

Can you check in that script in black-parrot and maintain it in sync with the source code?

dpetrisko, 2023-01-21:
Hi Alain, if you are pulling from master of BlackParrot, you'll need to update the filelist which changes as the code changes: https://github.com/black-parrot/black-parrot/blob/master/bp_top/syn/flist.vcs

For that specific file, I can tell you that the changeset is
-basejump_stl/bsg_cache/bsg_cache_sbuf_queue.v
+basejump_stl/bsg_cache/bsg_cache_sbuf.v
+basejump_stl/bsg_cache/bsg_cache_tbuf.v
+basejump_stl/bsg_cache/bsg_cache_buffer_queue.v

alaindargelas, 2023-01-21:
Hi Dan, I'd rather use the official flist.vcs then. How do I get the values of $BASEJUMP_STL_DIR and other variables?
Is there another step above that set that up?

dpetrisko, 2023-01-21:
Sounds good, easier to keep in sync that way.  We set those variables here: https://github.com/black-parrot/black-parrot/blob/master/Makefile.common#L1

```
export BP_COMMON_DIR    := $(TOP)/bp_common
export BP_FE_DIR        := $(TOP)/bp_fe
export BP_BE_DIR        := $(TOP)/bp_be
export BP_ME_DIR        := $(TOP)/bp_me
export BP_TOP_DIR       := $(TOP)/bp_top
export BP_EXTERNAL_DIR  := $(TOP)/external
export BASEJUMP_STL_DIR := $(BP_EXTERNAL_DIR)/basejump_stl
export HARDFLOAT_DIR    := $(BP_EXTERNAL_DIR)/HardFloat
```

alaindargelas, 2023-01-21:
Actually, this flow with flist.vcs is for simulation.
What I need is a deterministic way (In sync with the source code) to get the file list for Synthesis (The Surelog + Yosys flow we are trying to fix here).

dpetrisko, 2023-01-21:
This is the synth flist. There are some extra files which are used for other parameterizations of BP, but everything in there should be synth

alaindargelas, 2023-01-21:
ok, thanks

alaindargelas, 2023-01-21:
Since I want to debug the Surelog step stand alone first, I create the following script:

source black-parrot.sl

```
export TOP=/home/alain/black-parrot
export BP_COMMON_DIR=$TOP/bp_common
export BP_FE_DIR=$TOP/bp_fe
export BP_BE_DIR=$TOP/bp_be
export BP_ME_DIR=$TOP/bp_me
export BP_TOP_DIR=$TOP/bp_top
export BP_EXTERNAL_DIR=$TOP/external
export BASEJUMP_STL_DIR=$BP_EXTERNAL_DIR/basejump_stl
export HARDFLOAT_DIR=$BP_EXTERNAL_DIR/HardFloat

surelog -f bp_top/syn/flist.vcs
```
There are a couple of problems with -f in Surelog with the comments, so I removed them manually. That used to work but somehow got broken. I'll fix that.
And one file (bsg_circular_ptr.v) needs to -sv in front of it in the flist.vcs to be treated as a SV file.

I'll start by debugging the following errors:

```
[NTE:EL0531] /home/alain/black-parrot/external/HardFloat/source/HardFloat_primitives.v:63:17: Negative value in instance "work@addRecFN.addRecFNToRaw.lowMask_far_roundExtraMask"
             text:         input [(inWidth - 1):0] in,
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/bp_fe/src/v/bp_fe_icache.sv:269:9: Negative value in instance "work@bp_multicore.cc.y[0].x[0].tile_node.tile.core.core_lite.core_minimal.fe.icache"
             text:   wire [ctag_vbits_lp-1:0] ctag_vbits = vaddr_tl_r[block_offset_width_lp+sindex_width_lp+:`BSG_MAX(ctag_vbits_lp,1)];
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:343:9: Negative value in instance "work@bp_multicore.cc.y[0].x[0].tile_node.tile.core.core_lite.core_minimal.be.calculator.pipe_mem.dcache"
             text:   wire [ctag_vbits_lp-1:0] ctag_vbits = vaddr_tl_r[block_offset_width_lp+sindex_width_lp+:`BSG_MAX(ctag_vbits_lp,1)];
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/external/basejump_stl/bsg_noc/bsg_mesh_stitch.v:18:15: Negative value in instance "work@bp_multicore.ic.rev_mesh"
             text:    (input    [y_max_p-1:0][x_max_p-1:0][nets_p-1:0][S:W][width_p-1:0] outs_i // for each node, each direction
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/external/basejump_stl/bsg_noc/bsg_mesh_stitch.v:19:15: Negative value in instance "work@bp_multicore.ic.rev_mesh"
             text:     , output [y_max_p-1:0][x_max_p-1:0][nets_p-1:0][S:W][width_p-1:0] ins_o
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/external/basejump_stl/bsg_noc/bsg_mesh_stitch.v:22:20: Negative value in instance "work@bp_multicore.ic.rev_mesh"
             text:     , input  [E:W][y_max_p-1:0][nets_p-1:0][width_p-1:0] hor_i
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/external/basejump_stl/bsg_noc/bsg_mesh_stitch.v:23:20: Negative value in instance "work@bp_multicore.ic.rev_mesh"
             text:     , output [E:W][y_max_p-1:0][nets_p-1:0][width_p-1:0] hor_o
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/bp_fe/src/v/bp_fe_icache.sv:269:9: Negative value in instance "work@bp_unicore.unicore_lite.core_minimal.fe.icache"
             text:   wire [ctag_vbits_lp-1:0] ctag_vbits = vaddr_tl_r[block_offset_width_lp+sindex_width_lp+:`BSG_MAX(ctag_vbits_lp,1)];
             value: INT:-1.

[NTE:EL0531] /home/alain/black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:343:9: Negative value in instance "work@bp_unicore.unicore_lite.core_minimal.be.calculator.pipe_mem.dcache"
             text:   wire [ctag_vbits_lp-1:0] ctag_vbits = vaddr_tl_r[block_offset_width_lp+sindex_width_lp+:`BSG_MAX(ctag_vbits_lp,1)];
             value: INT:-1.

[NTE:EL0508] Nb Top level modules: 180.

[NTE:EL0509] Max instance depth: 24.

[NTE:EL0510] Nb instances: 5534.

[NTE:EL0511] Nb leaf instances: 924.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: /home/alain/black-parrot/slpp_all/surelog.uhdm ...

[INF:UH0709] Writing UHDM Html Coverage: /home/alain/black-parrot/slpp_all/checker/surelog.chk.html ...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 566
[   NOTE] : 336
```

I'm also looking at the html report for UHDM code coverage to inspect what parts of the design we are not covering. I'll need your help to review some too later on.

dpetrisko, 2023-01-21:
Sure thing, let me know. At a glance, those look like they are caused by default parameters of modules which are not in the elab tree for the default build