chipsalliance/synlig #1085: Formal equiv diff simple_signedexpr.v

alaindargelas, 2022-09-20:
Test yosys/tests/simple/signedexpr.v
./run_formal_verif.sh test=simple_signedexpr.v

Ignores the signeness of the port in the plugin AST

alaindargelas, 2022-09-21:
Debug technique used on that design:
./run_formal_verif.sh test=simple_signedexpr.v
DIFF

tkdiff build/tests/simple_signedexpr.v/yosys_gate.v build/tests/simple_signedexpr.v/surelog_gate.v 
Observe the netlist is drastically different

tkdiff build/tests/simple_signedexpr.v/yosys_ast.txt build/tests/simple_signedexpr.v/surelog_ast.txt 
Observe the AST is also drastically different

After commenting some of the assigns and seeing one is enough to have a diff, trim down the testcase to:
```
module signed_test01(a, b, yu);
input signed [1:0] a;
input signed [2:0] b;
output [3:0] yu;
assign yu = {a + b};
endmodule
```

Run again, save the results:
cp build/tests/simple_signedexpr.v/yosys_ast.txt build/tests/simple_signedexpr.v/yosys_ast.txt.signed
cp build/tests/simple_signedexpr.v/surelog_ast.txt build/tests/simple_signedexpr.v/surelog_ast.txt.signed

Change the testcase to:
```
module signed_test01(a, b, yu);
input [1:0] a;  // Remove the signed keyword
input signed [2:0] b;
output [3:0] yu;
assign yu = {a + b};
endmodule
```

Run again observe the test passes,

tkdiff build/tests/simple_signedexpr.v/yosys_ast.txt build/tests/simple_signedexpr.v/yosys_ast.txt.signed
Observe Yosys has a different AST, the signed variable is correctly annotated

tkdiff build/tests/simple_signedexpr.v/surelog_ast.txt build/tests/simple_signedexpr.v/surelog_ast.txt.signed
Observe the plugin AST is the same, the signed variable had no impact on the output.

Check UHDM, UHDM annotates the vpiSigned property to the typespec, conclude the signed information is not utilized in the plugin.

rkapuscik, 2022-12-30:
This is passing now: https://github.com/antmicro/yosys-systemverilog/blob/master/formal/passlist.txt#L262

closed 2022-12-30 by rkapuscik