chipsalliance/synlig #1137: Formal equiv diff simple_scopes.v

alaindargelas, 2022-10-09:
Test yosys/tests/simple_scopes.v
 ./run_formal_verif.sh test=simple_scopes.v

After fix: https://github.com/chipsalliance/Surelog/pull/3262 and https://github.com/chipsalliance/Surelog/pull/3263
One remaining issue in the plugin about the port  type of 'a' which is different from the Yosys AST version.
The result is a diffferent DSP setting in the gate level netlist. Yosys seems correct, and the plugin not.

[image]

alaindargelas, 2022-10-09:
[image]

wsipak, 2022-12-02:
Fixed here: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/408
And present in the passlist for CI testing here: https://github.com/antmicro/yosys-systemverilog/blob/568e88746964c87fd1e3495b729a6e0ebcf81fdc/formal/passlist.txt#L266

closed 2022-12-02 by wsipak