chipsalliance/synlig #1138: Formal equiv diff svtypes_typedef_memory_2.sv

alaindargelas, 2022-10-09:
Test yosys/tests/svtypes/typedef_memory_2.sv

The plugin erroneously creates some large constant (flattens the memory):

     AST_WIRE <typedef_memory_2.sv:4.0-4.0> [0x55b3136d7d60] str='\mem' logic basic_prep range=[3:0] multirange=[ 0 4 0 16 ] multirange_swapped=[ 0 1 ]
        ATTR \wiretype:
          AST_CONSTANT <typedef_memory_2.sv:0.0-0.0> [0x55b3136e52c0 -> 0x55b3136e8610] str='\nibble' bits='01011100011011100110100101100010011000100110110001100101'(56) basic_prep range=[55:0] int=1650617445
        ATTR \packed_ranges:

alaindargelas, 2023-12-11:
The plugin blasts the memory into bits which renders memory inference impossible.

kamilrakoczy, 2024-07-09:
`typedef_memory_2` test is now passing formal verification. Closing issue.

closed 2024-07-09 by kamilrakoczy