chipsalliance/synlig #1159: Formal equiv diff simple_operators.v

alaindargelas, 2022-10-16:
Test yosys/tests/simple/operators.v
After the signed problem get fixed, here are the next 2 problems with the testcase:

After reducing the test to:
```
module optest(clk, mode, u1, s1, u2, s2, y);
input clk;
input [6:0] mode;
input [3:0] u1, u2;
input [3:0] s1, s2;
output reg [7:0] y;
always @(posedge clk) begin
	y <= 8'h42;
	case (mode)
		40: y <= u1 + u2;
	endcase
end
endmodule
```
The gate level netlist contains a different order of the variable, it is formally the same but it looks like it might be source of the problem for the subsequent test:
```
module optest(clk, mode, u1, s1, u2, s2, y);
input clk;
input [6:0] mode;
input [3:0] u1, u2;
input [3:0] s1, s2;
output reg [7:0] y;
always @(posedge clk) begin
	y <= 8'h42;
	case (mode)
	        36: y <= u1 > u2;
		40: y <= u1 + u2;
	endcase
end
endmodule
```
The later test fails formal verification.

wsipak, 2023-10-04:
Unassigning myself as this is on hold.

alaindargelas, 2023-12-09:
Fixed

closed 2023-12-09 by alaindargelas