chipsalliance/synlig #1223: Formal equiv diff asicworld_code_hdl_models_arbiter_simple.v

alaindargelas, 2022-11-08:
Test yosys/tests/asicworld/code_hdl_models_arbiter_simple.v
UHDM dump seems all OK.
The simplified design below still exhibit the formal diff:

```
module arbiter (
  clk,    
  rst,    
  req3,   
  req2,   
  req1,   
  req0,   
  gnt3,   
  gnt1,   
  gnt0   
);
// --------------Port Declaration----------------------- 
input           clk;    
input           rst;    
input           req3;   
input           req2;   
input           req1;   
input           req0;   
output          gnt3;   
output          gnt1;   
output          gnt0;   

//--------------Internal Registers----------------------
wire    [1:0]   gnt       ;   
wire            beg       ;
wire   [1:0]    lgnt      ;
wire            lcomreq   ;
reg             lgnt0     ;
reg             lgnt1     ;
reg             lgnt2     ;
reg             lgnt3     ;
reg             lasmask   ;
reg             lmask0    ;
reg             lmask1    ;
reg             ledge     ;

//--------------Code Starts Here----------------------- 
always @ (posedge clk)
if (rst) begin
  lgnt0 <= 0;
  lgnt1 <= 0;
  lgnt2 <= 0;
  lgnt3 <= 0;
end else begin
   lgnt0 <=req0;
    lgnt1 <=( lmask1 );
    lgnt2 <= req0;
end 

//----------------------------------------------------
// lasmask state machine.
//----------------------------------------------------
assign beg = (req3 | req2 | req1 | req0) & ~lcomreq;
always @ (posedge clk)
begin                                     
  lasmask <= (beg & ~ledge & ~lasmask);
  ledge   <= (beg & ~ledge &  lasmask) 
          |  (beg &  ledge & ~lasmask);
end 

//----------------------------------------------------
// comreq logic.
//----------------------------------------------------
   
   assign lcomreq = ( req3 & lgnt3 ) 
      | ( req2 & lgnt2 )
      | ( req1 & lgnt1 ) 
      | ( req0 & lgnt0 );
   
//----------------------------------------------------
// Encoder logic.
//----------------------------------------------------
   assign  lgnt =  2'b11;
//----------------------------------------------------
// lmask register.
//----------------------------------------------------
always @ (posedge clk )
if( rst ) begin
  lmask1 <= 0;
  lmask0 <= 0;
end else if(lasmask) begin
   lmask1 <= 1'b1;
  lmask0 <= 1'b1; 
end
   
assign gnt    = lgnt;
//----------------------------------------------------
// Drive the outputs
//----------------------------------------------------
assign gnt1   = lgnt1;
assign gnt0   = lgnt0;

endmodule
```

wsipak, 2023-10-04:
Unassigning myself as this is on hold.

alaindargelas, 2023-12-09:
Fixed but Yosys formal fails to prove the equivalence for the original test, hence not in the passlist.
The simplified test passes.

closed 2023-12-09 by alaindargelas