chipsalliance/synlig #1248: Yosys `-formal` support

ollie-etl, 2022-11-14:
My appologies if i've missed this in the docs, but I'm unable to use as a front end for formal verification

This just fine: bmc, prove and cover statements all run ok
```console
read -sv -formal dataflow_formal.sv
read -sv -formal dataflow_id.sv
prep -top dataflow_id
```

This does not
```console
verilog_defines -DFORMAL=1 -USYNTHESIS
read_systemverilog -defer dataflow_formal.sv
read_systemverilog -defer dataflow_id.sv
prep -top dataflow_id
```

alaindargelas, 2022-11-15:
@tgorochowik can you check what is missing?

ollie-etl, 2022-11-17:
@rkapuscik Here is a reduced example

`ex1.sv`
```systemverilog
module ex1 (
    input  logic a,
    output logic b
);
  assign a = b;

  always_comb cover_in_eq_out : cover (a == b);
  always_comb cover_in_neq_out : cover (a != b);

  always_comb assert_a_eq_b : assert (a == b);
  always_comb assert_a_neq_b : assert (a != b);
endmodule
```

And the sby script `ex1.sby`
```
# See https://symbiyosys.readthedocs.io/en/latest/reference.html#
[tasks]
bmc
prove
cover

[options]
depth 2
append 1

bmc:   mode bmc
prove: mode prove
cover: mode cover

[engines]
smtbmc

[script]
# Replace `read` with `read_systemverilog`
# read -sv -formal ex1.sv
read_systemverilog ex1.sv
prep -top ex1

[files]
./ex1.sv
```

Extracts from `read -sv -formal` 
```log
SBY  8:57:53 [ex1_bmc] engine_0: smtbmc
SBY  8:57:53 [ex1_bmc] base: starting process "cd ex1_bmc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY  8:57:53 [ex1_bmc] base: finished (returncode=0)
SBY  8:57:53 [ex1_bmc] smt2: starting process "cd ex1_bmc/model; yosys -ql design_smt2.log design_smt2.ys"
SBY  8:57:53 [ex1_bmc] smt2: finished (returncode=0)
SBY  8:57:53 [ex1_bmc] engine_0: starting process "cd ex1_bmc; yosys-smtbmc --presat --unroll --noprogress -t 2  --append 1 --dump-vcd engine_0/trace.vcd --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Solver: yices
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Checking assumptions in step 0..
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Checking assertions in step 0..
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  BMC failed!
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Appending additional step 1.
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Re-solving with appended steps..
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Assert failed in ex1: assert_a_neq_b
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace.vcd
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace_tb.v
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace.smtc
SBY  8:57:53 [ex1_bmc] engine_0: ##   0:00:00  Status: failed
SBY  8:57:53 [ex1_bmc] engine_0: finished (returncode=1)
SBY  8:57:53 [ex1_bmc] engine_0: Status returned by engine: FAIL
SBY  8:57:53 [ex1_bmc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY  8:57:53 [ex1_bmc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
[... 185 lines cut ...]
SBY  8:58:05 [ex1_cover] base: Warning: ex1.sv:7: Skipping non-synthesizable object of type 'immediate_cover'
SBY  8:58:05 [ex1_cover] base: Warning: ex1.sv:8: Skipping non-synthesizable object of type 'immediate_cover'
SBY  8:58:05 [ex1_cover] base: finished (returncode=0)
SBY  8:58:05 [ex1_cover] smt2: starting process "cd ex1_cover/model; yosys -ql design_smt2.log design_smt2.ys"
SBY  8:58:05 [ex1_cover] smt2: finished (returncode=0)
SBY  8:58:05 [ex1_cover] engine_0: starting process "cd ex1_cover; yosys-smtbmc --presat --unroll -c --noprogress -t 2  --append 1 --dump-vcd engine_0/trace%.vcd --dump-vlogtb engine_0/trace%_tb.v --dump-smtc engine_0/trace%.smtc model/design_smt2.smt2"
SBY  8:58:05 [ex1_cover] engine_0: ##   0:00:00  Solver: yices
SBY  8:58:05 [ex1_cover] engine_0: ##   0:00:00  Status: passed
SBY  8:58:05 [ex1_cover] engine_0: finished (returncode=0)
SBY  8:58:05 [ex1_cover] engine_0: Status returned by engine: pass
SBY  8:58:05 [ex1_cover] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY  8:58:05 [ex1_cover] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY  8:58:05 [ex1_cover] summary: engine_0 (smtbmc) returned pass
SBY  8:58:05 [ex1_cover] DONE (PASS, rc=0)
SBY  8:58:05 The following tasks failed: ['bmc', 'prove']
```

This actually fairs better than my real example, in that it actually runs the asserts. Covers are stripped as non synthesisisable.
In reality, I instantiate a module within the design which contains the assumptions and assertions for the rules to apply for each port, as either a master or slave. I see warnings like

```
SBY  9:05:50 [dataflow_id_cover] base: Warning: Removing unused module: \dataflow_formal from the design.
```

I guess its stripped as all the ports are inputs. Is there a pragma to stop it stripping these, or a way when in formal mode to assume that code full of asserts / assumes is not unused?

ollie-etl, 2022-11-19:
A possibility 

```systemverilog

/// An interfaces for buses implementing dataflow handshake semantics
///
/// Invariants:
///
///   - A transaction occurs when both valid and ready are high
///   - A slave does not have to await valid before asserting ready
///   - Valid, once asserted must remain stable until ready is asserted
///   - `data` must be stable whilst valid is high
///   - Both `data` and `ready `may be undefined when valid is low
interface dataflow #(
    // The data type being transferred
    // parameter type data_t = logic

    // The width of the data bus being transferred
    parameter int DATA = 8
) (
    input logic clk,
    input logic areset
);

  logic valid;
  logic ready;
  logic [DATA-1:0] data;

  modport master(input clk, reset, ready, output valid, data);

  modport slave(input clk, reset, valid, data, output ready);

  modport observer(input clk, reset, valid, data, ready);

endinterface
```

```systemverilog
/*
  AMBA Axis

 Parameterized with marker types for determining if fields should be used or not
 This is as close to type safety as we can get, sadly.

 It has the nice effect of propagating generics into modules without requiring
 the modules themselves to be parameterized. If a modules requires a particular field
 that should be expressed as an initial assertion.
*/
interface axis #(
    // ************************
    // Bus field configuration
    // ************************

    /// If low, `tlast` is constant `1`
    parameter logic [0:0] HAS_LAST = 1,

    /// If low `tkeep` is considered all high (all bytes must be kept)
    parameter logic [0:0] HAS_KEEP = 0,

    /// If low `tstrb` is considered all high (all bytes are data)
    parameter logic [0:0] HAS_STRB = 0,

    /// If low, `tid` is constant `'0`
    parameter logic [0:0] HAS_ID = 1,

    /// If low, `tdest` is constant `'0`
    parameter logic [0:0] HAS_DEST = 1,

    /// If low, `tuser` is constant `'0`
    parameter logic [0:0] HAS_USER = 1,

    // ************************
    // Field configuration
    // ************************
    /// The width of the `tdata` field, in bytes
    parameter integer DATA = 8,

    /// The width of the `tdest` field, in bits
    parameter integer DEST = 8,

    /// The width of the `tid` field, in bits
    parameter integer ID = 8,

    // The width of the `tuser` field, in bits
    parameter integer USER = 8
) (
    input logic aclk,
    input logic aresetn
);

  // Indicates the bus has valid data. Data is stable whilst tvalid is high
  logic                   tvalid;

  // Indicates the slave has consumed the data this cycle.
  // A master must not await tready before asserting valid
  logic                   tready;

  // Indicates this is the last beat of a transfer
  // Designs may assume this is high is unused (i.e, single beat transactions)
  logic                   tlast;

  // A byte qualifier used to indicate whether the content of the associated byte must be transported to the destination.
  // Designs may assume this is high if unused (i.e, keep all bytes)
  logic [DATA-1 : 0]      tkeep;

  // A byte qualifier used to indicate whether the content of the associated byte is a data byte or a position byte.
  // Designs may assume this is high if unused (i.e, keep all bytes)
  logic [DATA-1 : 0]      tstrb;

  // The data stream identifier that indicates different streams of data.
  // Designs may assume this is zero if unused
  logic [  ID-1 : 0]      tid;

  // Routing information for the data stream.
  // Designs may assume this is zero if unused
  logic [  DEST-1:0]      tdest;

  // User defined sideband information that can be transmitted alongside the data stream.
  // Designs may assume this is zero if unused
  logic [  USER-1:0]      tuser;

  // TDATA is the primary payload that is used to provide the data that is passing across the interface. The width of the data payload is an integer number of bytes.
  logic [  DATA-1:0][7:0] tdata;

  modport master(
      input aclk, aresetn, tready,
      output tvalid, tlast, tstrb, tkeep, tid, tdest, tuser, tdata
  );

  modport slave(
      input aclk, aresetn, tvalid, tlast, tstrb, tkeep, tid, tdest, tuser, tdata,
      output tready
  );

  modport observer(
      input aclk, aresetn, tvalid, tlast, tstrb, tkeep, tid, tdest, tuser, tdata, tready
  );

  // **************************************
  // Runtime Axis configuration checks
  // **************************************
  initial begin : axis_optional_field_initial_properties
    // Assert these to get early failure in synthesis
    // Zero bit widths are not support in systemverilog
    non_zero_id : assert (ID > 0);
    non_zero_user : assert (USER > 0);
    non_zero_dest : assert (DEST > 0);
    non_zero_data : assert (DATA > 0);
  end

  // Assert bus invariants. These are generated on demand to improve simulation times
  generate
    // Assert unused bus fields behave as expected
    if (!HAS_LAST) always_comb if (tvalid) tlast_is_high_when_unused : assert (tlast);
    if (!HAS_KEEP) always_comb if (tvalid) tkeep_is_high_when_unused : assert (&tkeep);
    if (!HAS_STRB) always_comb if (tvalid) tstrb_is_high_when_unused : assert (&tstrb);
    if (!HAS_DEST) always_comb if (tvalid) tdest_is_zero_when_unused : assert (!|tdest);
    if (!HAS_USER) always_comb if (tvalid) tuser_is_zero_when_unused : assert (!|tuser);
    if (!HAS_ID) always_comb if (tvalid) tid_is_zero_when_unused : assert (!|tid);

    // Ensure these are never observed to be invalid
    if (HAS_KEEP && HAS_STRB)
      always_comb if (tvalid) tkeep_tstrb_are_valid : assert (!|(~tkeep & tstrb));

  endgenerate

endinterface

```

alaindargelas, 2022-11-22:
I made a fix in Surelog https://github.com/chipsalliance/Surelog/pull/3352 that will accept the -formal option and let the cover keyword be allowed in a "formal" category of keywords: synthesizable + formal.

ollie-etl, 2022-11-24:
@alaindargelas Thanks for looking into this.

alaindargelas, 2023-12-05:
The whole formal verification support in read_systemverilog is not going to get any attention until properly funded by someone.
A workaround, of course is to move those statements in files that can be read by read_verilog and read the rest of the design with read_systemverilog