chipsalliance/synlig #1448: bug with case..inside

astrollo, 2023-01-27:
It seems that the systemverilog plugin for yosys does not correctly handle the set membership case statement (standard sect. 12.5.4).  Consider the following simple listing:
```
module case_inside  (input logic [2:0] sel,   output logic [1:0] y);
always_comb
case (sel) inside
  3'b000, 3'b001, 3'b010, 3'b011 : y = 2'b01;
  [4:7] : y = 2'b10;
endcase
endmodule : case_inside
```
The second item of the case matches 'b100 'b101 'b110 'b111, while the first item matches the other four cases (thus synthesis after minimization should be y[1]=sel[2];  y[0]= ~sel[2])

I used the following script with yosys 0.25+86 (git sha1 541fdffff, clang 10.0.0-4ubuntu1 -fPIC -Os)
```
plugin -i systemverilog
read_systemverilog case_inside.sv
hierarchy -check 
proc
opt -full -purge -share_all
show
techmap
abc
clean
write_verilog -noattr
```
The show command before techmap already highlights that only the two values 'b100 'b111 are considered for the second item of the case. This is confirmed by synthesized verilog produced by yosys (with y[1] wrongly computed):
```
module case_inside(sel, y);
  wire _0_;
  input [2:0] sel;
  wire [2:0] sel;
  output [1:0] y;
  wire [1:0] y;
  assign y[0] = ~sel[2];
  assign _0_ = ~(sel[1] ^ sel[0]);
  assign y[1] = sel[2] & _0_;
endmodule
```

tgorochowik, 2023-01-31:
Thank you very much for the detailed report!

kamilrakoczy, 2023-02-06:
I've prepared PR that fixes this: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/450.
Using this PR I got:
```
module case_inside(sel, y);
  input [2:0] sel;
  wire [2:0] sel;
  output [1:0] y;
  wire [1:0] y;
  assign y[0] = ~sel[2];
  assign y[1] = sel[2];
endmodule
```

It adds only initial support for list op with low and high bound (``[4:7]``), so it may still not work in more complicated case.

closed 2023-02-08 by kamilrakoczy