chipsalliance/synlig #1450: bug with casez

astrollo, 2023-01-27:
It seems that the systemverilog plugin for yosys does not correctly handle casez. Consider the following simple listing:
```
module cz (input logic [2:0] s, output logic [1:0] y);
always_comb
 casez (s)
  3'b1?? : y = 2'b11;
  3'b01? : y = 2'b10;
  3'b001 : y = 2'b01;
  default : y= 2'b00;
 endcase
endmodule : cz
```

The expected result, after minimization, should be y[1]=s [2]^s[1]; y[0]=s [2] | (s[1] & ~s[0]) 

For synthesis I have used the following script with yosys 0.25+86 (git sha1 541fdffff, clang 10.0.0-4ubuntu1 -fPIC -Os)

```
plugin -i systemverilog
read_systemverilog cz.sv
hierarchy -check 
proc
clean
show
opt -full -purge -share_all
techmap
abc
clean
write_verilog -noattr
```

The output of the show command (before techmap) displays some $eq cells having constant inputs with Z values (hardly to understand from the synthesis point of view).
After techmap and simplification with abc the following description is obtained:

```
module cz(s, y);
  wire _0_;
  wire _1_;
  input [2:0] s;
  wire [2:0] s;
  output [1:0] y;
  wire [1:0] y;
  assign _0_ = s[0] ^ s[2];
  assign y[0] = _0_ & ~(s[1]);
  assign _1_ = s[1] ^ s[2];
  assign y[1] = _1_ & ~(s[0]);
endmodule
```
with inconsistent output.
Note that the output is instead correct if the input file is loaded in yosys without the systemverilog plugin, i.e. with:
`read_verilog -sv cz.sv`

tgorochowik, 2023-01-31:
Thank you very much for the detailed report!

kamilrakoczy, 2023-02-07:
I've created PR that fixes this: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/451
Using this PR I've got:
```
module top(s, y);                                                     
  wire _0_;                                                           
  input [2:0] s;                                                                                                                             
  wire [2:0] s;                                                       
  output [1:0] y;                                                                                                                                                                                                                                                                         
  wire [1:0] y;                                                                                                                              
  assign _0_ = s[0] & ~(s[1]);                                                                                                               
  assign y[0] = s[2] | _0_;                       
  assign y[1] = s[2] | s[1];                                                                                                                 
endmodule
```
It matches result from standard yosys without systemverilog plugin.

closed 2023-02-08 by kamilrakoczy