chipsalliance/synlig #1474: Can not initialize localparam array with function return value in SystemVerilog

forthyen, 2023-02-05:
Can not initialize localparam array with function return value in SystemVerilog.
yosys
plugin -i systemverilog
read_systemverilog test.sv
.....
[FTL:EL0546] /mnt/e/yosys-systemverilog/test.sv:13:3: Fatal elaboration "--[0] should be 5 ".

[FTL:EL0546] /mnt/e/yosys-systemverilog/test.sv:16:3: Fatal elaboration "--[1] should be 5".

`module top #(
)(
);
    typedef int unsigned ASSIGN_VADDR_RET_T[2];
    function static ASSIGN_VADDR_RET_T ASSIGN_VADDR();
        for (int i = 0; i < 2; i++) begin
            ASSIGN_VADDR[i] = 5;
        end
    endfunction

    localparam int unsigned VADDR = ASSIGN_VADDR();
	if (VADDR[0] != 5) begin
		$fatal(1,"--[0] should be 5 ");
	end;
	if (VADDR[1] != 5) begin
		$fatal(1,"--[1] should be 5");
	end;
endmodule

module main;
    top #() top1;
endmodule`

[test.zip](https://github.com/antmicro/yosys-systemverilog/files/10610127/test.zip)

kamilrakoczy, 2023-02-07:
Related to Surelog issue: https://github.com/chipsalliance/Surelog/issues/3460

forthyen, 2023-02-13:
Sorry! I post this bug onto "Surelog" in gitlab, and it is solved by downloading latest "Surelog" !
Thanks all !

closed 2023-02-13 by forthyen