chipsalliance/synlig #1533: Latest black-parrot with latest SURELOG macro support and latest plugin core dumps in RTLIL generation

alaindargelas, 2023-02-24:
cd black-parrot
yosys-uhdm-plugin-integration/image/bin/yosys -s yosys.ys
[yosys.ys.gz](https://github.com/antmicro/yosys-systemverilog/files/10820751/yosys.ys.gz)

```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.25+83 (git sha1 0d71dd316, gcc 11.2.0-7ubuntu2 -fPIC -Os)

-- Executing script file `yosys.ys' --

1. Executing Verilog with UHDM frontend.
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
....
Warning: Removing unused module: \bsg_mux_segmented__abstract from the design.
Generating RTLIL representation for module `$paramod$b97b6e11908ab2ff2b8d2f70f58c0d34d3473bc3\bsg_scan'.
Segmentation fault (core dumped)
```

alaindargelas, 2023-02-24:
Already visible in sv-tests:
https://chipsalliance.github.io/sv-tests-results/?v=uhdmyosys+black-parrot+bp_be

mglb, 2023-03-14:
The line causing the problem: https://github.com/bespoke-silicon-group/basejump_stl/blob/34d2e96d8649a502edae7577550845b02c19d3c0/bsg_misc/bsg_scan.v#L53

Minimal test case:
```systemverilog
module test_countones #( parameter val = 0) (input i, output logic o);
   initial assert($countones(val) == 1);
endmodule
```
The plugin ignores non-synthesizable objects from UHDM. In this case, the non-syntesizable object is `$countones(...)`. As a result, we end up with equivalent of `assert(/*...*/ == 1);` which is obviously invalid.

When `-synth` flag is passed to `read_systemverilog`, the problem is already detected by Surelog:
```
[ERR:UH0720] /home/user/p/repos/black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:1232:43: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/bp_me/src/v/cce/bp_cce_dir.sv:371:32: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/bp_me/src/v/cce/bp_cce_dir.sv:373:32: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_cache/bsg_cache.v:1191:16: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_cache/bsg_cache.v:1195:16: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_mem/bsg_cam_1r1w_tag_array.v:78:45: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_mem/bsg_cam_1r1w_tag_array.v:83:20: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_mem/bsg_mem_1r1w_one_hot.v:66:62: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_mem/bsg_mem_1r1w_one_hot.v:68:62: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_misc/bsg_reduce.v:20:15: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_misc/bsg_reduce_segmented.v:17:14: Non-synthesizable construct "$countones".

[ERR:UH0720] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_misc/bsg_scan.v:53:15: Non-synthesizable construct "$countones".
```

I wonder whether we should always enable `-synth` in the plugin instead of doing non-synthesizable objects removal ourselves?

Also, what do you think about support for `// synopsys translate_(on|off)` comments? Surelog already supports something like this, but with `//pragma translate_(on|off)` comments. This, together with `-synth`, could solve the problem of non-synthesizable SV code.

---

BTW. I've tried using `//pragma translate_(on|off)` comments in Black-Parrot, and seems like Surelog has some issues with it.

What I did:
- Executed in black-parrot directory: `find \( -name '*.v' -o -name '*.sv' \) -exec sed -i 's|// *synopsys  *translate_|//pragma translate_|' {} \;`
- Added `-synth` to `read_systemverilog` in `yosys.ys`
- Executed: `yosys -s ./yosys.ys`

Output:
```
-- Executing script file `./yosys.ys' --

1. Executing Verilog with UHDM frontend.
[SNT:PA0207] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_cache/bsg_cache_dma_to_wormhole.v:337: Syntax error: extraneous input '(' expecting {';', 'default', 'module', 'endmodule', 'extern', 'macromodule', 'interface', 'program', 'virtual', 'class', 'timeunit', 'timeprecision', 'checker', 'type', 'clocking', 'defparam', 'bind', 'const', 'function', 'static', 'constraint', 'if', 'automatic', 'localparam', 'parameter', 'specparam', 'import', 'genvar', 'typedef', 'enum', 'struct', 'union', 'string', 'chandle', 'event', '[', 'byte', 'shortint', 'int', 'longint', 'integer', 'time', 'bit', 'logic', 'reg', 'shortreal', 'real', 'realtime', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0', 'tri1', 'wire', 'uwire', 'wand', 'wor', 'trireg', 'signed', 'unsigned', 'interconnect', 'var', '$', 'export', DOLLAR_UNIT, '(*', 'assert', 'property', 'assume', 'cover', 'not', 'or', 'and', 'sequence', 'covergroup', 'pulldown', 'pullup', 'cmos', 'rcmos', 'bufif0', 'bufif1', 'notif0', 'notif1', 'nmos', 'pmos', 'rnmos', 'rpmos', 'nand', 'nor', 'xor', 'xnor', 'buf', 'tranif0', 'tranif1', 'rtranif1', 'rtranif0', 'tran', 'rtran', 'generate', 'case', 'for', 'global', 'initial', 'assign', 'alias', 'always', 'always_comb', 'always_latch', 'always_ff', 'restrict', 'let', 'this', 'randomize', 'final', 'task', 'specify', 'sample', '=', 'nettype', Escaped_identifier, Simple_identifier, '`pragma', SURELOG_MACRO_NOT_DEFINED},
($clog2(dma_burst_len_p+1+1))
^-- /home/user/p/repos/black-parrot/slpp_all/lib/work/external/basejump_stl/bsg_cache/bsg_cache_dma_to_wormhole.v:439:0:.

[SNT:PA0207] /home/user/p/repos/black-parrot/external/basejump_stl/bsg_cache/bsg_wormhole_to_cache_dma_fanout.v:413: Syntax error: extraneous input '(' expecting {';', 'default', 'module', 'endmodule', 'extern', 'macromodule', 'interface', 'program', 'virtual', 'class', 'timeunit', 'timeprecision', 'checker', 'type', 'clocking', 'defparam', 'bind', 'const', 'function', 'static', 'constraint', 'if', 'automatic', 'localparam', 'parameter', 'specparam', 'import', 'genvar', 'typedef', 'enum', 'struct', 'union', 'string', 'chandle', 'event', '[', 'byte', 'shortint', 'int', 'longint', 'integer', 'time', 'bit', 'logic', 'reg', 'shortreal', 'real', 'realtime', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0 [... cut ...] ', 'always_comb', 'always_latch', 'always_ff', 'restrict', 'let', 'this', 'randomize', 'final', 'task', 'specify', 'sample', '=', 'nettype', Escaped_identifier, Simple_identifier, '`pragma', SURELOG_MACRO_NOT_DEFINED},
( ((block_width_p*sets_p/8)==1) ? 1 : $clog2((block_width_p*sets_p/8)))
^-- /home/user/p/repos/black-parrot/slpp_all/lib/work/bp_fe/src/v/bp_fe_icache.sv:1079:0:.

[  FATAL] : 0
[ SYNTAX] : 6
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
ERROR: Error when parsing design. Aborting!
```

alaindargelas, 2023-03-15:
Surelog implements properly the synopsys pragma now with https://github.com/chipsalliance/Surelog/pull/3509
Please use the -synth option in sv-tests for black-parrot and update Surelog.

alaindargelas, 2023-03-15:
@mglb, this core dump is solved but there is another core dump in the plugin right after, in the same .v file. Please take a look at it.

closed 2023-11-14 by alaindargelas