chipsalliance/synlig #1575: Formal diff bsg_assembler_out_num_out_p_5_num_in_p_10_width_p_16_top.v (Plugin does not handle escaped names)

alaindargelas, 2023-03-12:
Test bsg_micro_designs/bsg_comm_link/bsg_assembler_out/bsg_assembler_out_num_out_p_5_num_in_p_10_width_p_16/top.v

First merge https://github.com/chipsalliance/Surelog/pull/3502 in the plugin, as there was a problem in Surelog for that test.
After merging and avoiding Surelog bug, the plugin truncates all escaped names.
In this testcase all escaped names are hierarchical names, maybe there is a problem with the . (hierarchical separator) in escaped names.
To be noted that in UHDM escaped names have been stripped of the leading "\" and trailing " " as is supposed to happen per standard.

mandrys, 2023-06-05:
Fix for escaped identifiers in the plugin has been merged: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/506.
However, _bsg_assembler_out_num_out_p_5_num_in_p_10_width_p_16_top.v_ test still doesn't pass the formal verification. We are looking into this issue, but at the moment we suspect that the formal equivalence check in yosys could be giving false results in this case.

alaindargelas, 2023-11-29:
The diffs are cosmetic only and Yosys formal engine is not strong enough to make the equivalence.
This is resolved.

closed 2023-11-29 by alaindargelas