chipsalliance/synlig #1587: Core dump Yosys AST bsg_strobe.json

alaindargelas, 2023-03-17:
Test bsg_micro_designs/bsg_misc/bsg_strobe/bsg_strobe.json
yosys-uhdm-plugin-integration$ ./run_formal_verif.sh test=bsg_strobe.json

cd bsg_micro_designs/bsg_misc/bsg_strobe
~/yosys-uhdm-plugin-integration/image/bin/yosys -s ~/yosys-uhdm-plugin-integration/build/tests/bsg_strobe.json/surelog.ys

```
555556f43ac0]
              AST_IDENTIFIER </home/alain/yosys-uhdm-plugin-integration/bsg_micro_designs/bsg_misc/bsg_strobe/src/bsg_reduce.v:27.18-27.19> [0x555556f43c80] str='\i'
--- END OF AST DUMP ---

Program received signal SIGSEGV, Segmentation fault.
0x00005555557e9b8c in Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) ()
(gdb) where
#0  0x00005555557e9b8c in Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) ()
...
```

wsipak, 2023-05-02:
With the current version of the plugin (`8bd3ecd`) and the `bsg_micro_designs` suite that is introduced here: https://github.com/antmicro/yosys-systemverilog/pull/1673, my observation is that the files in `bsg_strobe.json` test can be parsed in Yosys without errors.

Formal verification results in:
```
11. Executing EQUIV_MAKE pass (creating equiv checking module).
ERROR: Can't match gate port `init_val_r_i_gate' to a gold port.
```
However, I haven't encountered any segmentation fault when working with this test, so I expect the issue might have been solved already.

alaindargelas, 2023-05-02:
There is one core dump here:
https://chipsalliance.github.io/sv-tests-results/?v=uhdmyosys+black-parrot+bp_cce

probably the one in this case is gone.

closed 2023-12-07 by alaindargelas