chipsalliance/synlig #1766: 'plugin -i systemverilog' fails with

yurivict, 2023-06-06:
```
yosys> plugin -i systemverilog
src/tcmalloc.cc:333] Attempt to free invalid pointer 0x801b5d320

Program received signal SIGABRT, Aborted.
Sent by thr_kill() from pid 81993 and user 1001.
thr_kill () at thr_kill.S:4
4       in thr_kill.S
(gdb) bt
#0  thr_kill () at thr_kill.S:4
#1  0x00000008013719b4 in __raise (s=s@entry=6) at /disk-samsung/freebsd-src/lib/libc/gen/raise.c:52
#2  0x0000000801422359 in abort () at /disk-samsung/freebsd-src/lib/libc/stdlib/abort.c:67
#3  0x00000008034bd3cc in tcmalloc::Log(tcmalloc::LogMode, char const*, int, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem) () from /usr/local/lib/libtcmalloc.so.4
#4  0x00000008034b9979 in ?? () from /usr/local/lib/libtcmalloc.so.4
#5  0x00000008020c6b55 in systemverilog_plugin::UhdmCommonFrontend::UhdmCommonFrontend(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits
<char>, std::__1::allocator<char> >) () from /usr/local/bin/../share/yosys/plugins/systemverilog.so
#6  0x00000008020cb5d6 in systemverilog_plugin::UhdmSurelogAstFrontend::UhdmSurelogAstFrontend() () from /usr/local/bin/../share/yosys/plugins/systemverilog.so
#7  0x00000008020cb4d3 in ?? () from /usr/local/bin/../share/yosys/plugins/systemverilog.so
#8  0x0000000800eb433d in objlist_call_init (list=list@entry=0x7fffffffda00, lockstate=lockstate@entry=0x7fffffffd978) at /disk-samsung/freebsd-src/libexec/rtld-elf/rtld.c:3135
#9  0x0000000800eb8a6c in dlopen_object (name=name@entry=0x801b3a180 "/usr/local/bin/../share/yosys/plugins/systemverilog.so", fd=fd@entry=-1, refobj=<optimized out>, lo_flags=<optimized out>, mode=mode@entry=1,
    lockstate=0x7fffffffd978, lockstate@entry=0x0) at /disk-samsung/freebsd-src/libexec/rtld-elf/rtld.c:3883
#10 0x0000000800eb530e in rtld_dlopen (name=0x801b3a180 "/usr/local/bin/../share/yosys/plugins/systemverilog.so", fd=-1, mode=<optimized out>) at /disk-samsung/freebsd-src/libexec/rtld-elf/rtld.c:3743
#11 0x00000000008502b7 in Yosys::load_plugin(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocat
or<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >) ()
#12 0x00000000008508b0 in Yosys::PluginPass::execute(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_tra
its<char>, std::__1::allocator<char> > > >, Yosys::RTLIL::Design*) ()
#13 0x00000000005a8278 in Yosys::Pass::call(Yosys::RTLIL::Design*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std:
:__1::char_traits<char>, std::__1::allocator<char> > > >) ()
#14 0x00000000005a7ee0 in Yosys::Pass::call(Yosys::RTLIL::Design*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) ()
#15 0x0000000000639fee in Yosys::shell(Yosys::RTLIL::Design*) ()
#16 0x000000000059ca93 in main ()
(gdb) 

```
yosys-0.29
yosys-systemverilog-2023-06-01
clang-14
FreeBSD 13.2

yurivict, 2023-06-07:
The problem was due to surelog using the libtcmalloc library.
Plugins can't use different memory allocation libraries.

closed 2023-06-07 by yurivict

hzeller, 2023-06-07:
Interesting, so that would mean for a distribution of the surelog libraries, tcmalloc should not be used so that it is compatible with whatever else is out there (maybe libsurelog as shared object would work ?).

While a binary such as Surelog itself can choose tcmalloc or not as it is self-contained. This should be explored.

yurivict, 2023-06-07:
Or Yosys itself can be made to link with tcmalloc. This is what I did in FreeBSD ports: I added the TCMALLOC option to all relevant ports. In order for yosys-systemverilog to work all TCMALLOC options should be either ON or OFF.

The way how it is now - when only Surelog links with tcmalloc by default - will cause problems in general.