chipsalliance/synlig #1776: Can't load module `./systemverilog': /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found

gkamendje, 2023-06-09:
I am currently running Ubuntu 22.04 LTS with  Yosys 0.13+15 (git sha1 bc027b2ca ) installed with OpenRoad Flow Scripts
When following the instruction to download and install the Systemverilog module, 
```
curl https://api.github.com/repos/antmicro/yosys-systemverilog/releases/latest | jq .assets[1] | grep "browser_download_url" | grep -Eo 'https://[^\"]*' | xargs wget -O - | tar -xz
```
Looks like the latest version downloaded seems to be tailored only for Ubuntu 20.04 
`https://github.com/antmicro/yosys-systemverilog/releases/download/b9f2e2b-2023-06-07/yosys-uhdm-plugin-b9f2e2b-Ubuntu-20.04-focal-x86_64.tar.gz`
As a matter of fact after using the installation script to install the plugin in the Yosys directory, I get the following message
```
# This [00000.000000] ERROR: Can't load module `./systemverilog': /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/yosys/bin/../share/yosys/plugins/systemverilog.so)
```
Libc6 is already installed on my system. 
Is there a way to successfully install the SystemVerilog plugin on Ubuntu 22.04 LTS.

tgorochowik, 2023-06-26:
Hmm, most of the CI already runs on ubuntu 22.04, we switched to it as it's the latest LTS. Only the release is still done on 20.04 - I think we should just bump this

tgorochowik, 2023-06-26:
See: #1816

It looks as though currently we're preparing the packages on Ubuntu 22.04 and only release on 20.04  - that means that the name of the artifact was wrong but the files should work on 22.04, so this PR might not be enough to fix this issue

mglb, 2023-06-26:
@gkamendje please run following commands and post their output:
```
YOSYS_BIN=/home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
file $YOSYS_BIN
ldd $YOSYS_BIN
```
Adjust the path to yosys if I get it wrong (I've deduced it from log you posted).
`file` prints short information about file, I am interested in binary type (static or shared linking, etc)
`ldd` prints list of shared libraries used by the binary. I am interested whether it uses standard system libraries or comes with its own.

gkamendje, 2023-06-27:
I think I made a mistake when I opened this issue. I am currently running on  `Ubuntu 20.04.6 LTS` which explain why the mislabelled 22.04 version will not run on my system as explained in #1816. 
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
```
@mglb here are the output of `file` and `ldd`
```
$ file /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
/home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=bdced67d593b0328ea00877bd6e8c8bb429be1ea, for GNU/Linux 3.2.0, not stripped
```
```
$ ldd /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
        linux-vdso.so.1 (0x00007ffe6ecc5000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd4b9a06000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd4b98b7000)
        libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007fd4b9867000)
        libffi.so.7 => /usr/lib/x86_64-linux-gnu/libffi.so.7 (0x00007fd4b985b000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd4b9855000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd4b9839000)
        libtcl8.6.so => /usr/lib/x86_64-linux-gnu/libtcl8.6.so (0x00007fd4b967b000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd4b9660000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd4b946e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd4ba524000)
        libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fd4b943e000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd4b941b000)
```
If i understand #1816 properly, newer releases will only run on `Ubuntu 22.04` is that correct? Given that there are still many tools such as Yosys that rely on older version of glibc wouldn't it be a good idea to continue to generate releases for ``Ubuntu 20.04`?

mglb, 2023-06-29:
> I think I made a mistake when I opened this issue. I am currently running on `Ubuntu 20.04.6 LTS` which explain why the mislabelled 22.04 version will not run on my system as explained in #1816.

OK, this explains everything.

> If i understand #1816 properly, newer releases will only run on `Ubuntu 22.04` is that correct?

Yes.

> Given that there are still many tools such as Yosys that rely on older version of glibc wouldn't it be a good idea to continue to generate releases for ``Ubuntu 20.04`?

We plan to try static linking of a libc with the plugin. This should allow use of the plugin on wide range of distributions.

gkamendje, 2023-07-04:
I updated my WSL env to Ubuntu 22.04. Everything works fine. Closing this issue

tgorochowik, 2023-07-05:
Thank you for the confirmation!

closed 2023-07-05 by tgorochowik

Andrea-Spitale, 2023-07-16:
Hi,
I face the very same issue. How did you solve it? Is it just a matter of updating glibc? I am using a dockerized version of openlane, which includes yosis, hence updating glibc is not really straightforward, and I am running Ubuntu 22.04 thourhg windows subsystem for linux.

Thanks!

gkamendje, 2023-07-16:
@Andrea-Spitale  In my case, updating to Ubuntu 22.04 solved the issue because I was running a version intended for Ubuntu 22.04 on a Ubuntu 20.04 machine. Not sure how things work in a docker container environment.

Andrea-Spitale, 2023-07-17:
@gkamendje so you're using the so called "containerless" version of openlane? i.e. you installed all the tools manually, etc
Does everything work fine for you? If so, I could consider doing that

gkamendje, 2023-07-17:
@Andrea-Spitale I am not using OpenLane, I am using a local installation of OpenRoadFlowScripts