chipsalliance/synlig #1791: command terminated by singal 11

gkamendje, 2023-06-15:
I am trying to modify the Openroad Flow scripts to support the systemverilog module. I thought it would be just as simple as replacing 
```
foreach file $::env(VERILOG_FILES) {
  read_verilog -defer -sv {*}$vIdirsArgs $file
}
```
with something like
```
foreach file $::env(VERILOG_FILES) {
  read_systemverilog -defer -verbose -d 2 {*}$vIdirsArgs  -sv  $file
}
```
and properly loading the module with `-m systemverilog`. However this systematically fails after reading the second SystemVerilog file with a reason that I do not understand. Here are last lines that are printed out.
```
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 1
[   NOTE] : 5
Command terminated by signal 11
```
The log file `slpp_all/surelog.log` does not contain any suspicious information.
Synthesis runs to completion if instead I do pass the SystemVerilog files as follows. 
```
read_systemverilog -defer -verbose -d 2 {*}$vIdirsArgs  -sv  my_path_to_file1.sv \
                                                                                                my_path_to_file2.sv \
                                                                                                my_path_to_file3.sv \
                                                                                                my_path_to_file4.sv 
```
which indicates that the SystemVerilog files are all ok.

tgorochowik, 2023-06-26:
The `-defer` flag is currently experimental and might not work in all cases. Is it possible to share a design on which this happens? (preferable something small that reproduces this issue).

gkamendje, 2023-09-04:
The following works well for me. I was missing the `-link` after reading with the `-defer` option 
```
if {[info exist ::env(YOSYS_READ_SV)]} {
  
   foreach file $::env(VERILOG_FILES) {
      puts "reading file  $file"
      read_systemverilog -defer  -verbose -d 2 {*}$vIdirsArgs -DSYNTHESIS -sv $file
   }
   read_systemverilog -link -verbose -d 2
}
```

closed 2023-09-04 by gkamendje