chipsalliance/synlig #1798: Hashes in tag names prevent auto-discovery of new versions

yurivict, 2023-06-18:
The system compares hashes as versions:
[image]

This prevents auto-discovery in this particular case on FreeBSD, but the same problem exists on all major distros.

Please consider to either move the hash to the end of the tag, or to remove it completely.

Thank you,
Yuri (maintainer of the FreeBSD port)

tgorochowik, 2023-06-26:
No objections from me, see #1815 

We need to move the hash in case there are two releases in the same day. Will this not be problem btw? i.e. two releases in a day.

We could add some kind of a patch level to the tag, or maybe switch to a complete timestamp?

tgorochowik, 2023-06-26:
we could use this:
```
git show -s --date=short --format=%cd-%h --date=format:%Y%m%d%H%M%S
20230626103106-be367c6
```
@yurivict would this work for you?

mglb, 2023-06-26:
Reopening due to question above.

PR #1815 (2023-06-26): Update tags/releases format
Moving date to the beginning of the tag makes it easier to sort and detect changes.

Closes: #1798
commit cbf68daf (2023-06-26): Update tags/releases format (#1815)

Moving date to the beginning of the tag makes it easier to sort and
detect changes.

Closes: #1798