chipsalliance/synlig #1832: glibc version mismatch

Andrea-Spitale, 2023-07-16:
Hi,
I face the very same issue as https://github.com/antmicro/yosys-systemverilog/issues/1776.

 How did you solve it? Is it just a matter of updating glibc? I am using a dockerized version of openlane, which includes yosis, hence updating glibc is not really straightforward, and I am running Ubuntu 22.04 thourhg windows subsystem for linux.

Thanks!

tgorochowik, 2023-07-19:
Hi, we're working on improving the releases we produce to get rid of the dependencies on a specific GLIBC version

Andrea-Spitale, 2023-07-19:
Hi @tgorochowik 
Thanks for your reply. I'll wait then :)

wormyrocks, 2023-08-06:
FWIW, I got around this by installing oss-cad-suite and copying the yosys binary from the antmicro release over the one from oss-cad-suite. Then installing the systemverilog plugin as usual.

gmlarumbe, 2023-08-26:
> FWIW, I got around this by installing oss-cad-suite and copying the yosys binary from the antmicro release over the one from oss-cad-suite. Then installing the systemverilog plugin as usual.

This one worked for me. However with this workaround the ghdl plugin bundled with `oss-cad-suite` will not load anymore:
```
yosys> plugin -i ghdl
ERROR: Can't load module `./ghdl': libffi.so.7: cannot open shared object file: No such file or directory
```

Seyviour, 2023-12-13:
> FWIW, I got around this by installing oss-cad-suite and copying the yosys binary from the antmicro release over the one from oss-cad-suite. Then installing the systemverilog plugin as usual.

Hello. I have the GLIBC mismatch problem running yosys from `oss-cad-suite`. I've tried searching for the antmicro yosys release but I can't seem to find it;  could you kindly point me to where I could get it?