chipsalliance/synlig #1843: abort in yosys frontends/ast/genrtlil.cc:705 when using systemverilog plugin

mayagokhale, 2023-07-25:
Using yosys 0.30+16 (git sha1 8b2a00102, gcc 11.3.0-1ubuntu1~22.04.1 -fPIC -Os) built from master branch of https://github.com/antmicro/yosys-systemverilog
 
Error message is
Generating RTLIL representation for module `\encode_stream_sc_module_8'.
ERROR: Abort in frontends/ast/genrtlil.cc:705.

file attached. It's a system verilog file, but renamed to .txt suffix so it would upload.
[encode_stream_error.txt](https://github.com/antmicro/yosys-systemverilog/files/12163848/encode_stream_error.txt)

tgorochowik, 2023-07-27:
Thanks for the report! Could you try to minimize the provided systemverilog sample to something minimal? The approach that usually works best is removing parts of the code that seem unrelated until the error stops appearing. This should allow you to the lines that cause this issue

mayagokhale, 2023-07-27:
@tgorochowik Thank you for taking a look at the example file. The original file has 100K+ lines. This is the only module out of 19 in that design that fails. I think the problem is the functions declared in this module., since that is the only module that has functions. 

I tried running yosys under gdb to dump out the AST node it's failing on, but the binary didn't include symbols, and the AST dump at that default case at line 705 was commented out.

 Interestingly, the command 'read_verilog' rather than 'read_systemverilog' in yosys runs to completion. 
When I have more time, I'll rebuild yosys to dump the ast node that isn't being handled.

alaindargelas, 2023-10-16:
The example below reproduces the probem.
Basically, the plugin does not support raw function calls in statements, it does support them when they are part of a rhs assign though.

```
module top(input logic clk);
logic a;
logic b;
  always @(posedge clk) begin: mc_proc
    sc_dtsc_min(a, a);
  end
  
  function automatic logic  sc_dtsc_min (input logic a, input logic b);
  begin
     sc_dtsc_min = (a) <= (b) ? a : b;
  end
  endfunction
endmodule
```

alaindargelas, 2023-10-20:
@mayagokhale , I just ran the latest Yosys that is merged here in the plugin using the **read_verilog** command on your original design,
I get this error:

 ERROR: Can't resolve task name `\zhwencode_streamfp_t11_52_2_bits_t64pack_bits'.

It looks like this is a Yosys problem, not a plugin problem, Yosys cannot handle this raw function call. It works if you transform the statement as an assignment and make the function call the rhs of the assign.

```
if ((ts_local_0_bits__ref_0) < (minbits)) begin
          pbits_local_5 = (minbits) - (ts_local_0_bits__ref_0);
          abits_local_6 = (64) - ((ts_local_0_bits__ref_0) & ((64) - (1)));
          n_local_7 = sc_dtsc_min(pbits_local_5,abits_local_6);
          debug = zhwencode_streamfp_t11_52_2_bits_t64pack_bits(ts_local_0_s__ref_0,ts_local_0_prec__ref_0,ts_local_0_planes__ref_0,ts_local_0_buf__ref_0,ts_local_0_bits__ref_0,ts_local_0_posr__ref_0,ts_local_0_posw__ref_0,n_local_7,64'd0,ts_local_0_s__ref_0,ts_local_0_prec__ref_0,ts_local_0_planes__ref_0,ts_local_0_buf__ref_0,ts_local_0_bits__ref_0,ts_local_0_posr__ref_0,ts_local_0_posw__ref_0);
        end
```
Can you double check and rerun you case?
Please file an issue to Yosys if you see the same error as I see.
I'll close this issue here, please re-open if you have a different outcome.

closed 2023-10-20 by alaindargelas