chipsalliance/synlig #1859: Types defined in a package have incorrect width

gdnmhr, 2023-08-07:
I found some strange behavior when using types defined in a package. I have the following structure:

`my_pgk.sv`:

    package my_pkg;
    
      typedef logic [31:0] my_type_t;
    
    endpackage

`module.sv`:

    module top ();
    
    my_pkg::my_type_t my_wire;
    logic [31:0] my_wire_2;
    
    endmodule

If I try to import it into yosys using the following commands:

    plugin -i systemverilog
    read_systemverilog my_pkg.sv module.sv
    write_verilog output.v

Then I get the following output:

    /* Generated by Yosys 0.30+48 (git sha1 b5b0b7e83, clang 14.0.0-1ubuntu1 -fPIC -Os) */
    
    (* cells_not_processed =  1  *)
    (* src = "/home/yosys/debug_type/module.sv:1.1-6.10" *)
    module top();
      (* src = "/home/yosys/debug_type/module.sv:3.19-3.26" *)
      (* wiretype = "\\my_type_t" *)
      wire [1023:0] my_wire;
      (* src = "/home/yosys/debug_type/module.sv:4.14-4.23" *)
      wire [31:0] my_wire_2;
    endmodule

I would expect the width of wire `my_wire` to be 32, not 1024.

tgorochowik, 2023-08-08:
Thanks for the report!

alaindargelas, 2023-08-29:
The UHDM dump is correct:

```
|uhdmtopModules:
\_module_inst: work@top (work@top), id:42, file:/home/alain/Surelog/tests/ScratchPad.sv, line:9:1, endln:14:10
  |vpiName:work@top
  |vpiVariables:
  \_logic_var: (work@top.my_wire), id:25, line:11:19, endln:11:26
    |vpiParent:
    \_module_inst: work@top (work@top), id:42, file:/home/alain/Surelog/tests/ScratchPad.sv, line:9:1, endln:14:10
    |vpiTypespec:
    \_logic_typespec: (my_type_t), id:20, line:4:11, endln:4:23
      |vpiName:my_type_t
      |vpiInstance:
      \_package: my_pkg (my_pkg::), id:1, file:/home/alain/Surelog/tests/ScratchPad.sv, line:2:1, endln:6:11
      |vpiRange:
      \_range: , id:21, line:4:17, endln:4:23
        |vpiParent:
        \_logic_typespec: (my_type_t), id:20, line:4:11, endln:4:23
        |vpiLeftRange:
        \_constant: , id:22, line:4:18, endln:4:20
          |vpiParent:
          \_range: , id:21, line:4:17, endln:4:23
          |vpiDecompile:31
          |vpiSize:64
          |UINT:31
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , id:23, line:4:21, endln:4:22
          |vpiParent:
          \_range: , id:21, line:4:17, endln:4:23
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiName:my_wire
    |vpiFullName:work@top.my_wire
    |vpiVisibility:1
  |vpiVariables:
  \_logic_var: (work@top.my_wire_2), id:33, line:12:14, endln:12:23
    |vpiParent:
    \_module_inst: work@top (work@top), id:42, file:/home/alain/Surelog/tests/ScratchPad.sv, line:9:1, endln:14:10
    |vpiTypespec:
    \_logic_typespec: , id:29, line:12:1, endln:12:13
      |vpiParent:
      \_logic_var: (work@top.my_wire_2), id:33, line:12:14, endln:12:23
      |vpiRange:
      \_range: , id:26, line:12:7, endln:12:13
        |vpiParent:
        \_logic_typespec: , id:29, line:12:1, endln:12:13
        |vpiLeftRange:
        \_constant: , id:27, line:12:8, endln:12:10
          |vpiParent:
          \_range: , id:26, line:12:7, endln:12:13
          |vpiDecompile:31
          |vpiSize:64
          |UINT:31
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , id:28, line:12:11, endln:12:12
          |vpiParent:
          \_range: , id:26, line:12:7, endln:12:13
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiName:my_wire_2
    |vpiFullName:work@top.my_wire_2
    |vpiVisibility:1
    |vpiRange:
    \_range: , id:30, line:12:7, endln:12:13
      |vpiParent:
      \_logic_var: (work@top.my_wire_2), id:33, line:12:14, endln:12:23
      |vpiLeftRange:
      \_constant: , id:31, line:12:8, endln:12:10
        |vpiParent:
        \_range: , id:30, line:12:7, endln:12:13
        |vpiDecompile:31
        |vpiSize:64
        |UINT:31
        |vpiConstType:9
      |vpiRightRange:
      \_constant: , id:32, line:12:11, endln:12:12
        |vpiParent:
        \_range: , id:30, line:12:7, endln:12:13
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
```

alaindargelas, 2023-09-13:
Actually, the UHDM tree repeats the same information twice, 32 bit variable with a 32 bit wide typespec which leads to a 1024 wide net.

alaindargelas, 2023-09-13:
The fix in surelog is just a cleanup, it actually removes a repeat of the range for the variable my_wire2. 
It does not fix this case for my_wire.

The problem happens in uhdm_ast::add_multirange_wire , it replicates the range of the typespec and attaches it as an attribute where the net object already has the typespec as a children and the typespec already has the range.
[image]

@tgorochowik @kamilrakoczy FYI, this is a pretty serious bug that we will encounter often.

alaindargelas, 2023-10-26:
Actually fixed by #2072

closed 2023-10-26 by alaindargelas