chipsalliance/synlig #1908: Encountered unhandled object 'xxx' of type 'method_func_call'

renu883, 2023-09-01:
steps to reproduce (blanking out function name):
read_systemverilog -defer f1.sv
read_systemverilog -defer f2.sv
...
read_systemverilog -defer f12.sv
read_systemverilog -link

hierarchy -check -top top

ERROR: Encountered unhandled object 'xxx' of type 'method_func_call'

alaindargelas, 2023-09-12:
Please provide an actual testcase (SystemVerilog file).

alaindargelas, 2023-10-25:
Thanks for the testcase.
The plugin does not support SystemVerilog Interfaces **at all** at this point. 
So even if we fixed that particular error, there would be dozen more things to fix afterwards.
I'm afraid there is no way forward here. Unless you want to fund the development for SV Interfaces.

renu883, 2026-05-16:
Thank you so much Alain.