chipsalliance/synlig #1934: undefined symbol: _ZN5Yosys8Frontend7executeESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EEPNS_5RTLIL6DesignE

gkamendje, 2023-09-06:
Using version 3d041dfdcc81082ae19af8a22a01200387747baa 
OS= CentOS Linux release 7.9.2009
I am running Centos7 and I was able to compile and install as reported here #1826. However, I get the message `undefined symbol: _ZN5Yosys8Frontend7executeESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EEPNS_5RTLIL6DesignE`
when  I try to load the module in Yosys.
```
Yosys 0.13+15 (git sha1 bc027b2ca, gcc 8.3.1 -fPIC -Os)

ERROR: Can't load module `./systemverilog': /my_path/OpenROAD-flow-scripts/tools/install/yosys/bin/../share/yosys/plugins/systemverilog.so: undefined symbol: _ZN5Yosys8Frontend7executeESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EEPNS_5RTLIL6DesignE

```
Any hint what is going on?

pgielda, 2023-09-14:
It means that the plugin is referencing to _ZN5Yosys8Frontend7executeESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EEPNS_5RTLIL6DesignE symbol that is not in the yosys binary.

```
echo "_ZN5Yosys8Frontend7executeESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EEPNS_5RTLIL6DesignE" | c++filt 
Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*)
```
Most probably it means that the signature changed.

The pinned version to this repo is currently https://github.com/YosysHQ/yosys/tree/8b2a0010216f9a15c09bd2f4dc63691949b126df which is newer (0.33+34) so I would suggest using a newer yosys version.

gkamendje, 2023-09-15:
I get the following error message while trying to use it with the latest version of Yosys (b84ed5d3ad2ac8b533e1d6a83dbdb6d5c2be9ce7). I am running `./build_binaries.sh -y`. My version of the plugin is  bc99d465403d663bffa9f91d36e3b16a5cc79a6c
```
+ make -C /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog install -j8
make: Entering directory `/home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog'
mkdir -p /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin
PKG_CONFIG_PATH=/home/new_user/tmp_compile/UHDM/synlig/image/lib/pkgconfig:/cad/verilator/v5.002/share/pkgconfig PKG_CONFIG_PATH_FOR_TARGET=/home/new_user/tmp_compile/UHDM/synlig/image/lib/pkgconfig: pkg-config --cflags Surelog
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/uhdm_ast.cc.o /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_ast.cc
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/uhdm_ast_frontend.cc.o /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_ast_frontend.cc
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/uhdm_common_frontend.cc.o /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_common_frontend.cc
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/uhdm_surelog_ast_frontend.cc.o /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_surelog_ast_frontend.cc
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/const2ast.cc.o /home/new_user/tmp_compile/UHDM/synlig/third_party/yosys_mod/const2ast.cc
-I/home/new_user/tmp_compile/UHDM/synlig/image/include  
g++ -I../../third_party/yosys_mod -Wall -Wextra -ggdb -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/share/yosys/include -MD -MP -D_YOSYS_ -fPIC -I~/tmp_compile/openroad-flow-branch-smic018/tools/install/yosys/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER   -std=c++17 -Wall -W -Wextra -Wno-deprecated-declarations -Wno-unused-parameter -I/home/new_user/tmp_compile/UHDM/synlig/image/include     -MMD -c -o /home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/simplify.cc.o /home/new_user/tmp_compile/UHDM/synlig/third_party/yosys_mod/simplify.cc
In file included from /home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_surelog_ast_frontend.cc:20:
/home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_ast.h:4:10: fatal error: frontends/ast/ast.h: No such file or directory
[... cut ...]
/home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog/uhdm_ast.h:4:10: fatal error: frontends/ast/ast.h: No such file or directory
    4 | #include "frontends/ast/ast.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [/home/new_user/tmp_compile/UHDM/synlig/build/systemverilog-plugin/uhdm_ast_frontend.cc.o] Error 1
make: Leaving directory `/home/new_user/tmp_compile/UHDM/synlig/frontends/systemverilog'

```
I do see the header files in the include subdirectory of the Yosys installation directory. Any idea what is going on?

gkamendje, 2023-09-15:
looks like the issue is reported in #1961

pgielda, 2023-09-24:
Yes, it will be fixed via #2008

alaindargelas, 2023-10-16:
This is fixed with the latest version.

closed 2023-10-16 by alaindargelas