chipsalliance/synlig #1961: latest yosys: ast process_format_str() changed, plugin won't compile.

hzeller, 2023-09-12:
Looks like the new function is now called processFormat(). So the systemverilog plug--in will not compile with latest yosys from head as our local copy of `simplify.cc` uses that in [two](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/128ed9ce9cafc1a95b26f29a98c76c1e520451a6/systemverilog-plugin/third_party/yosys/simplify.cc#L728) [places](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/128ed9ce9cafc1a95b26f29a98c76c1e520451a6/systemverilog-plugin/third_party/yosys/simplify.cc#L3334). So probably some `#ifdef` of sorts needed to work with older and newer yosys-es.

(this is copied from https://github.com/chipsalliance/yosys-f4pga-plugins/issues/542 )

hzeller, 2023-09-15:
The [currently referenced yosys](https://github.com/YosysHQ/yosys/tree/8b2a0010216f9a15c09bd2f4dc63691949b126df) is more than 3 months old, so there will be a problem compiling the plugin with folks using newer Yosys versions.

Making it compatible with older and newer Yoysys API using a `#ifdef` or entirely implement the function in question independently of Yosys can be the way forward @wsipak

timkpaine, 2023-09-15:
https://github.com/YosysHQ/yosys/pull/3721

alaindargelas, 2023-10-13:
Latest Yosys is merged. https://github.com/chipsalliance/synlig/pull/2063

closed 2023-10-13 by alaindargelas