chipsalliance/synlig #1962: The plugin is cloning too many repos (linux!?)

alain-rs, 2023-09-13:
Especially for users, all tests subsystems should be separated out. 

systemverilog-plugin$ git submodule update --init --recursive
Submodule 'uhdm-tests/black-parrot/black-parrot' (https://github.com/black-parrot/black-parrot.git) registered for path 'third_party/black_parrot'
Submodule 'uhdm-tests/black-parrot/black-parrot-sdk' (https://github.com/black-parrot-sdk/black-parrot-sdk.git) registered for path 'third_party/black_parrot_sdk'
Submodule 'uhdm-tests/black-parrot/black-parrot-tools' (https://github.com/black-parrot/black-parrot-tools) registered for path 'third_party/black_parrot_tools'
Submodule 'uhdm-tests/bsg_micro_designs/bsg_micro_designs' (https://github.com/bsg-idea/bsg_micro_designs.git) registered for path 'third_party/bsg_micro_designs'
Submodule 'third_party/googletest' (https://github.com/google/googletest) registered for path 'third_party/googletest'
Submodule 'uhdm-tests/ibex/ibex' (https://github.com/lowRISC/ibex.git) registered for path 'third_party/ibex'
Submodule 'make-env' (https://github.com/SymbiFlow/make-env.git) registered for path 'third_party/make_env'
Submodule 'uhdm-tests/opentitan/opentitan' (https://github.com/antmicro/opentitan.git) registered for path 'third_party/opentitan'
Submodule 'uhdm-tests/opentitan/opentitan-9d82960888' (https://github.com/lowRISC/opentitan.git) registered for path 'third_party/opentitan_9d82960888'
Submodule 'sv2v' (https://github.com/zachjs/sv2v.git) registered for path 'third_party/sv2v'
Submodule 'third_party/veer' (https://github.com/chipsalliance/Cores-VeeR-EH1) registered for path 'third_party/veer'
Submodule 'yosys-f4pga-plugins' (https://github.com/chipsalliance/yosys-f4pga-plugins.git) registered for path 'third_party/yosys_f4pga_plugins'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_tools'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/bsg_micro_designs'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/googletest'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/ibex'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/make_env'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/opentitan'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/opentitan_9d82960888'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/sv2v'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/veer'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/yosys_f4pga_plugins'...
Submodule path 'third_party/black_parrot': checked out '669bf7e5796d26d524a345510de4947742c6afe1'
Submodule 'external/HardFloat' (https://github.com/bsg-external/HardFloat.git) registered for path 'third_party/black_parrot/external/HardFloat'
Submodule 'external/basejump_stl' (https://github.com/bespoke-silicon-group/basejump_stl.git) registered for path 'third_party/black_parrot/external/basejump_stl'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot/external/HardFloat'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot/external/basejump_stl'...
Submodule path 'third_party/black_parrot/external/HardFloat': checked out '55b17279663ef1809f98c8c37f229454d76fb067'
Submodule path 'third_party/black_parrot/external/basejump_stl': checked out 'e90a34120b43c4ab3a8ee0b5bca1cebccb2eee1c'
Submodule 'imports/DRAMSim3' (https://github.com/bsg-external/DRAMsim3) registered for path 'third_party/black_parrot/external/basejump_stl/imports/DRAMSim3'
Submodule 'imports/ramulator' (https://github.com/bespoke-silicon-group/ramulator) registered for path 'third_party/black_parrot/external/basejump_stl/imports/ramulator'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot/external/basejump_stl/imports/DRAMSim3'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot/external/basejump_stl/imports/ramulator'...
Submodule path 'third_party/black_parrot/external/basejump_stl/imports/DRAMSim3': checked out '7630b9eb0e1ad60df8d52c185af292ea98c806ad'
Submodule 'ext/SuperLU_MT_3.1' (https://github.com/umd-memsys/SuperLU_MT_3.1.git) registered for path 'third_party/black_parrot/external/basejump_stl/imports/DRAMSim3/ext/SuperLU_MT_3.1'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot/external/basejump_stl/imports/DRAMSim3/ext/SuperLU_MT_3.1'...
Submodule path 'third_party/black_parrot/external/basejump_stl/imports/DRAMSim3/ext/SuperLU_MT_3.1': checked out '68739c5a23cc48c9a13d28793834e4fee95d33ee'
Submodule path 'third_party/black_parrot/external/basejump_stl/imports/ramulator': checked out '3582c445bab9b6be52e126338f8ca676f4fc2e34'
Submodule path 'third_party/black_parrot_sdk': checked out 'f014649b23ddc5715f619bf4fcbdd56105726e50'
Submodule 'bedrock' (https://github.com/black-parrot-sdk/bedrock.git) registered for path 'third_party/black_parrot_sdk/bedrock'
Submodule 'beebs' (https://github.com/black-parrot-sdk/beebs) registered for path 'third_party/black_parrot_sdk/beebs'
Submodule 'bootrom' (https://github.com/black-parrot-sdk/bootrom) registered for path 'third_party/black_parrot_sdk/bootrom'
Submodule 'bp-demos' (https://github.com/black-parrot-sdk/bp-demos) registered for path 'third_party/black_parrot_sdk/bp-demos'
Submodule 'bp-tests' (https://github.com/black-parrot-sdk/bp-tests) registered for path 'third_party/black_parrot_sdk/bp-tests'
Submodule 'coremark' (https://github.com/black-parrot-sdk/coremark) registered for path 'third_party/black_parrot_sdk/coremark'
Submodule 'dromajo' (https://github.com/bsg-external/dromajo) registered for path 'third_party/black_parrot_sdk/dromajo'
Submodule 'linux' (https://github.com/black-parrot-sdk/bp-linux) registered for path 'third_party/black_parrot_sdk/linux'
Submodule 'opensbi' (https://github.com/black-parrot-sdk/opensbi) registered for path 'third_party/black_parrot_sdk/opensbi'
Submodule 'perch' (https://github.com/black-parrot-sdk/perch) registered for path 'third_party/black_parrot_sdk/perch'
Submodule 'riscv-dv' (https://github.com/black-parrot-sdk/riscv-dv) registered for path 'third_party/black_parrot_sdk/riscv-dv'
Submodule 'riscv-gnu-toolchain' (https://github.com/black-parrot-sdk/riscv-gnu-toolchain) registered for path 'third_party/black_parrot_sdk/riscv-gnu-toolchain'
Submodule 'riscv-opcodes' (https://github.com/riscv/riscv-opcodes.git) registered for path 'third_party/black_parrot_sdk/riscv-opcodes'
Submodule 'riscv-tests' (https://github.com/black-parrot-sdk/riscv-tests) registered for path 'third_party/black_parrot_sdk/riscv-tests'
Submodule 'spec2000' (https://github.com/black-parrot-sdk/spec2000) registered for path 'third_party/black_parrot_sdk/spec2000'
Submodule 'spec2006' (https://github.com/black-parrot-sdk/spec2006.git) registered for path 'third_party/black_parrot_sdk/spec2006'
Submodule 'spec2017' (https://github.com/black-parrot-sdk/spec2017.git) registered for path 'third_party/black_parrot_sdk/spec2017'
Submodule 'yocto' (https://github.com/black-parrot-sdk/bp-yocto.git) registered for path 'third_party/black_parrot_sdk/yocto'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/bedrock'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/beebs'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/bootrom'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/bp-demos'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/bp-tests'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/coremark'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/dromajo'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/linux'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/opensbi'...
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/perch'...
[... cut ...]
Submodule path 'third_party/black_parrot_sdk/linux': checked out '4f35d5d83a61c4c785549895508e1365a5ebfe32'
Submodule 'buildroot' (https://github.com/buildroot/buildroot.git) registered for path 'third_party/black_parrot_sdk/linux/buildroot'
Submodule 'linux' (https://github.com/torvalds/linux.git) registered for path 'third_party/black_parrot_sdk/linux/linux'
Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/linux/buildroot'...
**Cloning into '/home/alain/RapidSilicon/systemverilog-plugin/third_party/black_parrot_sdk/linux/linux'...**

pgielda, 2023-09-14:
for now probably the way to go is to not clone all the submodules but just the ones needed for compilation. I do agree that we have to deal with it at some point, but currently we're trying to get a monolithic version with all the dependencies in one place and once we get that working we can discuss on how to trim this.

I would actually suggest (@tgorochowik @kgugala) for all the dependencies that are just needed for CI tests to be stored *not* as submodules, but somewhere in test scripts / CI scripts... Unless we're using some kind of a bot to bump that automatically and that is the reason?

timkpaine, 2023-09-15:
https://github.com/chipsalliance/UHDM/issues/1000 and https://github.com/chipsalliance/Surelog/issues/3845 may help, they've been on my backlog for a while

I have all the code to build .deb packages from Surelog / UHDM / Yosys, maybe we can piggy back on that. With `ccache` on GitHub actions, it's also completely viable to just build and install everything in the action (as we do for both surelog and UHDM, e.g. https://github.com/chipsalliance/Surelog/blob/master/.github/workflows/non_vendored.yml#L44). It is slow the first time but quite fast on subsequent runs.

mglb, 2023-09-19:
> I would actually suggest (@tgorochowik @kgugala) for all the dependencies that are just needed for CI tests to be stored _not_ as submodules, but somewhere in test scripts / CI scripts... Unless we're using some kind of a bot to bump that automatically and that is the reason?

We do use dependabot for some, e.g. opentitan.
[`submodule.<name>.update = none`](https://git-scm.com/docs/git-submodule#Documentation/git-submodule.txt-update--init--remote-N--no-fetch--no-recommend-shallow-f--force--checkout--rebase--merge--referenceltrepositorygt--depthltdepthgt--recursive--jobsltngt--no-single-branch--filterltfilterspecgt--ltpathgt82308203) submodule config option could probably be used to disable non-essential modules from being automatically fetched.

PR #2012 (2023-09-26): Do not pull non-essential submodules by default.
Only yosys and surelog submodules are needed for plugin build. Those two are automatically fetched as before, i.e. after calling just `git submodule update --recursive`.

All other submodules are skipped, even when their paths are explicitly specified. To fetch them one has to add `--checkout` flag has to `git submodule update`, e.g.:

```
git submodule update --dept 1 --init --recursive --checkout
```

Fixes #1962
commit af866852 (2023-09-26): Do not pull non-essential submodules by default. (#2012)

Only yosys and surelog submodules are needed for plugin build. Those two
are automatically fetched as before, i.e. after calling just `git
submodule update --recursive`.

All other submodules are skipped, even when their paths are explicitly
specified. To fetch them one has to add `--checkout` flag has to `git
submodule update`, e.g.:

```
git submodule update --dept 1 --init --recursive --checkout
```

Fixes #1962
closed 2023-09-26 by kbieganski