chipsalliance/synlig #1963: Assert `AST_INTERNAL::current_scope.count(wiretype_node->str)' failed in /home/alain/RapidSilicon/systemverilog-plugin/frontends/systemverilog/uhdm_ast.cc:466.

alain-rs, 2023-09-13:
There seems to be a problem in setting up the current_scope, the top level typespec are not part of the scope when processing the module M port declaration.

https://github.com/chipsalliance/synlig/blob/4fb4ad4dd4b4ae52fd70df117b944c04a4778a82/frontends/systemverilog/uhdm_ast.cc#L2089

```
typedef struct packed {
  logic [7:0] prd_1;
  logic [3:0] prd_2;
  logic [7:0] prd_3;
  logic [7:0] prd_4;
} prd_in_t;

module M(input prd_in_t     prd_i);  //  << Problem is here
endmodule

module top();
  prd_in_t    in_prd;  //  << No problem here

  assign in_prd = '{prd_1: 8'b0,  prd_2: 4'b1,  prd_3: 8'b0,  prd_4: 8'b0};    //  << No problem here

M m  (.prd_i ( in_prd  ));

endmodule
```

alaindargelas, 2023-09-13:
@kamilrakoczy @tgorochowik this is a problem showing in quite a few designs.

kbieganski, 2023-09-27:
@alaindargelas This seems to be working for me now, can you confirm?

alaindargelas, 2023-09-27:
@kbieganski this error is gone in the original testcase.
A new error is showing up:
ERROR: Assert `wire_node->attributes.count(ID::wiretype)' failed in /home/alain/RapidSilicon/Raptor_Tools/parser_plugins/synlig_plugin/frontends/systemverilog/uhdm_ast.cc:482

I'll need to extract a small example.
Closing this case.

alaindargelas, 2023-09-29:
This attached testcase has the same assert as the original case.

[kmac.tar.gz](https://github.com/chipsalliance/synlig/files/12756352/kmac.tar.gz)

ERROR: Assert `AST_INTERNAL::current_scope.count(wiretype_node->str)' failed in /home/alain/synlig/frontends/systemverilog/uhdm_ast.cc:466

kbieganski, 2023-10-05:
Not right now. We're focusing on general stability and usability at this time.

alaindargelas, 2023-10-13:
Multiplied define package (test.sv redefined the incomplete package kmac_pkg), Surelog needs a better filter to parse -y files.

alaindargelas, 2023-10-13:
https://github.com/chipsalliance/Surelog/issues/3902

closed 2023-10-13 by alaindargelas

alaindargelas, 2023-10-13:
Fixed by https://github.com/chipsalliance/Surelog/pull/3903