chipsalliance/synlig #1980: Support and test Mac build and install

timkpaine, 2023-09-15:
xref: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/485

aWZHY0yQH81uOYvH, 2023-11-07:
I'm still running into problems building on macOS. On the current head commit (b746752):

 * Makefile does not work, requiring many GNU-specific extensions (for `make`, `bash`, `date`, etc...)
 * CMake-based build almost works, getting to the final linking step for `systemverilog.dylib`, but cannot find symbols from Yosys.

<details>
<summary>Partial linker output</summary>

```
[100%] Linking CXX shared library /opt/synlig/install/share/yosys/plugins/systemverilog.dylib
ld: warning: ignoring duplicate libraries: 'third_party/surelog/third_party/UHDM/lib/libuhdm.a'
ld: Undefined symbols:
  Yosys::log_header(Yosys::RTLIL::Design*, char const*, ...), referenced from:
      systemverilog_plugin::UhdmAstFrontend::call_log_header(Yosys::RTLIL::Design*) in uhdm_ast_frontend.cc.o
      systemverilog_plugin::UhdmSurelogAstFrontend::call_log_header(Yosys::RTLIL::Design*) in uhdm_surelog_ast_frontend.cc.o
      systemverilog_plugin::SynligEdifBackend::execute(std::__1::basic_ostream<char, std::__1::char_traits<char>>*&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, Yosys::RTLIL::Design*) in synlig_edif.cc.o
  Yosys::log_signal(Yosys::RTLIL::SigSpec const&, bool), referenced from:
      systemverilog_plugin::SynligEdifBackend::execute(std::__1::basic_ostream<char, std::__1::char_traits<char>>*&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, Yosys::RTLIL::Design*) in synlig_edif.cc.o
      systemverilog_plugin::SynligEdifBackend::execute(std::__1::basic_ostream<char, std::__1::char_traits<char>>*&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, Yosys::RTLIL::Design*) in synlig_edif.cc.o
      systemverilog_plugin::SynligEdifBackend::execute(std::__1::basic_ostream<char, std::__1::char_traits<char>>*&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, Yosys::RTLIL::Design*) in synlig_edif.cc.o
      systemverilog_plugin::SynligEdifBackend::execute(std::__1::basic_ostream<char, std::__1::char_traits<char>>*&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, Yosys::RTLIL::Design*) in synlig_edif.cc.o
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
  Yosys::logv_error(char const*, __va_list_tag*), referenced from:
      systemverilog_plugin::UhdmAst::report_error(char const*, ...) const in uhdm_ast.cc.o
  Yosys::log_warning(char const*, ...), referenced from:
      systemverilog_plugin::UhdmAst::process_object(unsigned int*) in uhdm_ast.cc.o
      systemverilog_plugin::UhdmAst::process_design() in uhdm_ast.cc.o
      systemverilog_plugin::UhdmAst::process_operation(UHDM::BaseClass const*) in uhdm_ast.cc.o
      systemverilog_plugin::UhdmAst::process_operation(UHDM::BaseClass const*) in uhdm_ast.cc.o
      systemverilog_plugin::UhdmAst::process_unsupported_stmt(UHDM::BaseClass const*, bool) in uhdm_ast.cc.o
      systemverilog_plugin::const2ast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, char, bool) in synlig_const2ast.cc.o
      my_strtobin(std::__1::vector<Yosys::RTLIL::State, std::__1::allocator<Yosys::RTLIL::State>>&, char const*, int, int, char, bool) in synlig_const2ast.cc.o
      ...
  Yosys::AST_INTERNAL::current_ast, referenced from:
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
  Yosys::AST_INTERNAL::flag_mem2reg, referenced from:
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
  Yosys::AST_INTERNAL::current_block, referenced from:
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
      systemverilog_plugin::simplify(Yosys::AST::AstNode*, bool, bool, bool, int, int, bool, bool) in synlig_simplify.cc.o
[... cut ...]
third_party/surelog/third_party/antlr4/runtime/Cpp/runtime/libantlr4-runtime.a
-Xlinker
-framework
-Xlinker
CoreFoundation
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/lib/libz.tbd
-ldl
-lutil
-lm
-lpthread
```

</details>

 * CMake flags: none (using vendored Yosys)
 * CMake version: 3.27.7
 * clang: Apple clang version 15.0.0 (clang-1500.0.40.1) Target: x86_64-apple-darwin23.1.0

Following the same steps under Ubuntu 22.04 and GCC, the build runs successfully.

timkpaine, 2023-11-07:
this will work to manually install them, it should also work with uhdm and surelog from homebrew https://github.com/dau-dev/tools

alaindargelas, 2023-11-14:
Fixed by: #2133

make -f cmake-makefile

closed 2023-11-14 by alaindargelas