chipsalliance/synlig #1981: Add instructions and tests for non-vendored thirdparty build/install

timkpaine, 2023-09-15:
E.g. I would like to use Surelog and UHDM from my environment against a tarball (non git) version of this repo, as is required by many registries

kbieganski, 2023-09-26:
I think this will require adjustments to the build process, as the plugin build invokes Surelog installation in the output directory. But we should do it in the new makefiles from https://github.com/chipsalliance/synlig/pull/1941

As an aside, how viable is it to use the plugin with a non-vendored Surelog? From what I've seen the versions are pretty tightly coupled.

timkpaine, 2023-09-26:
It's not too bad to do: https://github.com/dau-dev/tools/blob/5bdc4cbc469a9195875c10a5c3cfbc503b93ebb9/Makefile#L382

it is required for many registries that you e.g. work from a tarball, don't vendor your dependencies, etc, so it should be a supported pattern even if the process requires a specific release of Surelog (but if you have Surelog as a git submodule, you should be pinning to a release anyway, not just "master").