chipsalliance/synlig #1982: Do not iterate vpiAllModule, vpiAllPackage anymore

alaindargelas, 2023-09-15:
As far as synthesis subset is concerned, the plugin should not iterate vpiAllModule, vpiAllPackage anymore.
All the information necessary to synthesize the design is in vpiTopModule, vpiTopPackage.

This is the root cause of the issue and attempted fix here:

https://github.com/chipsalliance/UHDM/pull/1019

This is not the right fix. The right fix is to iterate the vpiTopModule, vpiTopPackage.

alaindargelas, 2023-09-16:
Addendum: Surelog also now filters out the vpiAllModules that are not used in the design elaboration when -synth option is used, which is the case here.
So I merged the fix https://github.com/chipsalliance/UHDM/pull/1019. 

But please consider removing all traversal to vpiAllModule, vpiAllPackage, they should not be needed anymore and will potentially lead to other problems if kept.

wsipak, 2023-10-04:
Unassigning myself as this is on hold.

closed 2023-11-10 by alaindargelas