chipsalliance/synlig #2021: ERROR: Assert `wire_node->attributes.count(ID::wiretype)'

alaindargelas, 2023-09-28:
ERROR: Assert `wire_node->attributes.count(ID::wiretype)' failed in uhdm_ast.cc:482.

[wiretype.tar.gz](https://github.com/chipsalliance/synlig/files/12744641/wiretype.tar.gz)

I cannot reduce the test, I tried, sub pieces of the testcase seems to work fine.
It looks like a lot of definitions are involved. This needs some better debug dumping procedure to figure out what is involved in this assert.
Is the assert genuine, also, might be in question.

kbieganski, 2023-09-28:
Minimized the test:

```systemverilog
typedef enum logic { FOO } foo;

module dut(output foo [0:0] bar[1]);
endmodule

module top;
  foo [0:0] bar [1];
  dut u_dut (.*);
endmodule 
```

kbieganski, 2023-09-29:
Seems like the output enum wire doesn't get flagged as `is_custom_type` for some reason, so it doesn't pass [this check](https://github.com/chipsalliance/synlig/blob/aeba30fb35b78dfabe708d6ca1e3488b2e5fb614/third_party/yosys_mod/simplify.cc#L1491). This results in the wiretype not being set. Not sure why it happens, but it may be due to some simplification process (the wire has an `is_simplified_wire` attribute at this point).

alaindargelas, 2023-10-09:
@kbieganski , PR needs review:
https://github.com/chipsalliance/synlig/pull/2052

closed 2023-10-13 by alaindargelas