chipsalliance/synlig #2049: systemverilog-plugin fails with ERROR: Assert `did_insert' failed in .../UhdmAst.cc:611

MikeLWPh, 2023-10-05:
I am trying to synthesize a module using `Yosys 0.30+16 (git sha1 8b2a00102, gcc 9.3.1 -fPIC -Os)` with the systemverilog plugin from `synlig-plugin-bb5d911-2023-10-02.tar.gz`, but it fails in the `read_systemverilog -link` command. Here are the last few lines of output:

```
Warning: /home/mike/chips/digital/eic/rtl/pam4_mlse_va/rtl/pam4_mlse_va_mstep.sv:1051: Post-incrementation operations are handled as pre-incrementation.
Warning: /home/mike/chips/digital/eic/rtl/pam4_mlse_va/rtl/pam4_mlse_va_mstep.sv:1056: Post-incrementation operations are handled as pre-incrementation.
ERROR: Assert `did_insert' failed in /home/jamespires/git/yosys-systemverilog/yosys-f4pga-plugins/systemverilog-plugin/UhdmAst.cc:611.
make: *** [yosys] Error 1
```

I have tried chopping out parts of the RTL, but have not managed to locate the part that causes the problem. It looks like a bug in the plugin to me. Does anyone know how to fix this or find a workaround?

MikeLWPh, 2023-10-06:
@kgugala, thanks for your reply. Yes, sorry, that log was from an earlier version. I have now built synlig and yosys from the 2023-10-04 release and I get a very similar message:
```ERROR: Assert `did_insert' failed in /home/mike/Tools/git/synlig/frontends/systemverilog/uhdm_ast.cc:565.```
I have been hacking away at the original RTL, and I have managed to produce a very minimal version which shows the bug, which is attached below. Having reduced it so much, it looks like the problem is caused by the function returning a 2D array. Presumably synlig should accept this? It would be good to fix it, but in the meantime I will see if I can rewrite the RTL to work around it.

[bug_did_insert.sv.txt](https://github.com/chipsalliance/synlig/files/12834482/bug_did_insert.sv.txt)

alaindargelas, 2023-10-26:
Fixed by #2103

closed 2023-10-26 by alaindargelas

MikeLWPh, 2023-11-02:
Hi Alain. Thanks for that fix, but I'm struggling to try it out. On 27 Sept I managed to clone synlig and build synlig/surelog/yosys using build_binaries.sh. However when I updated my workspace to get your fix, it also got the New build setup (#1941) and I can't get this to work. Our system is RedHat CentOS Linux release 7.7.1908 (Core) which was released 2019-09-17. It seems to have several features missing which are now required to build synlig, e.g. env -S. I have tried asking our sysadmin if it could be updated to a newer version, but that's not likely to happen soon. I have also tried installing in my area several packages:  coreutils-9.0, bash-5.2.15, make-4.4.1, and cmake-3.27.7, and this has got me a bit further through the build process, but it's still broken. Is it possible to revert synlig to the old build setup?

At the moment it fails with the following output:
cd /home/mike/Tools/git/synlig/third_party/yosys/
make PREFIX:=/home/mike/Tools/git/synlig/out/release/ CONFIG:=clang CC:=cc CXX:=g++ -std=gnu++11 LD:=g++ -std=gnu++11  --no-print-directory
make: invalid option -- '='
make: invalid option -- 'g'
make: invalid option -- 'u'
make: invalid option -- '+'
etc.

So it seems the CXX and LD arguments have been split at the space, so the second part is treated as make flags, rather than part of the variable value. I can't see how to fix this. Any ideas?

alaindargelas, 2023-11-02:
@MikeLWPh, try the alternative build system instead. I checked in last week:
make -f cmake-makefile

Make sure you have cmake 3.20 at minimum.
That build system works on CentOS.

MikeLWPh, 2023-11-10:
Thanks for that tip. I have now managed to build the latest version of synlig/surelog/yosys, and I can confirm that the did_insert error is fixed, so the simple test case I sent now synthesises ok. However the real RTL still gives errors. On various different runs with different RTL yosys aborted with the following errors:
ERROR: Assert `!wire->name.empty()' failed in kernel/rtlil.cc:2111.
ERROR: Abort in frontends/ast/genrtlil.cc:786.
ERROR: :0: Encountered unhandled typespec in process_typespec_member: '' of type 'array_typespec'
Do any of these look like bugs you might already be working on? Or do I need to submit a simple testcase for each one?

alaindargelas, 2023-11-10:
Please file issues for each separate cases. Cases might look alike but they generally are not.