chipsalliance/synlig #2053: Simple interface throws warning, generates wrong code

mole99, 2023-10-08:
First of all, thank you all for your great work!

I tried to process a simple interface:

```SystemVerilog
interface bus_if;
    logic out;
    modport master(output out);
endinterface

module bus_master (bus_if.master bus, input logic din);
    assign bus.out = din;
endmodule

module top (input logic din);
    bus_if bus();
    bus_master m0(bus, din);
endmodule
```

In the stage `Generating RTLIL representation for module `\bus_master'.` I get the warning:

`Warning: Identifier `\bus.out' is implicitly declared.`

The output that is written by Yosys is not correct, `bus` is `inout` and there is no output for `bus.out`.

```Verilog
/* Generated by Yosys 0.33+65 (git sha1 3319fdc46, g++ 12.2.0-14 -fPIC -Os) */

module bus_if();
endmodule

module bus_master(bus, din);
  inout bus;
  wire bus;
  wire \bus.out ;
  input din;
  wire din;
  assign \bus.out  = din;
endmodule

module top(din);
  input din;
  wire din;
endmodule
```

alaindargelas, 2023-10-11:
@mole99 , the plugin does not support SystemVerilog interfaces. Surelog (the parser) does. But a critical piece is missing here.

alaindargelas, 2023-10-11:
Yosys obviously does not support them either. The plugin should transform them into modules/structs.

mole99, 2023-10-11:
I see, thanks for the clarification! Yes, Yosys does not support them, that's why it would be great if synlig does one day :D